rusefi/rusefi #3350: external muxes for internal ADC

rusefillc, 2021-10-13:
Ethernet pin conflict raised the issue of ADC shortage

At least on STM32H7 we can use external muxes to multiply number of channels by N

@mck1117 says

the plan on atlas is to use external muxes to multiply number of channels by N
sample group A
flip mux
sample group B
flip mux
sample group A

mck1117, 2021-10-13:
The H7's ADC is fast enough to do that flip/sample/flip/sample *for the fast ADC*.

mck1117, 2021-10-13:
Do we need to on F4? What's wrong with the setup today on F4?

rusefillc, 2021-10-13:
Ethernet on f4 seems to be asking

rusefillc, 2022-06-28:
8chan as first board with mux?

https://github.com/rusefi/alphax-8chan/issues/3

rusefillc, 2023-01-08:
Ugly limited implementation works

linked tune & log uses muxed analog inputs for pedal, and non-muxed input mux

https://rusefi.com/online/view.php?msq=1356
https://rusefi.com/online/view.php?log=1136

closed 2023-05-24 by rusefillc

dron0gus, 2024-07-20:
Some fixes https://github.com/rusefi/rusefi/pull/6732