rusefi/rusefi #3367: I BROKE SOMETHING?!

rusefillc, 2021-10-19:
very unexpected diff

looks like depending on sequence or SOMETHING we get random bundle info in "default" bundle?

[image]

https://github.com/rusefi/rusefi/commit/a1ed2018779e483f249cf4c72fed2711b03fb9e0

rusefillc, 2021-10-19:
@chuckwagoncomputing looks like this is about partial pinout push

whatever pinout file is the only single one to be update overrides generated files for "main" bundle

this is probably related to the fact that we use same path and at least some work flow depends on "default" bundle being handles _last_

in case of partial incremental generation that last default is skipped leaving whatever pinout was touched in default generated files

chuckwagoncomputing, 2021-11-23:
If I'm understanding correctly, the easy way to fix this would be to reset all non-board-specific files before running the generation for default. That way if the default hasn't changed, it'll leave the files unchanged.
This solution seems kind of ugly to me, but I'm not sure how to do it better.
Maybe we could remove all non-board-specific file paths from gen_config_common.sh and gen_config_board.sh, and give gen_config_default.sh it's own java call instead of just calling gen_config_board.sh in it, but then we introduce more code duplication, which is something we're trying to reduce in the build scripts.

PR #5021 (2023-02-01): Reset missed default config file
Fix #3367
closed 2023-02-01 by rusefillc