rusefi/rusefi #3395: Timestamp Error

BeerMoneyMotorsports, 2021-10-23:
I have experienced this along with Hellen users.  Can we fix this annoyance?  Complete power cycle resolves the issue.  

1-started the engine
2-stopped the engine without powering down ECU
3-waited minutes
4-attempted to start engine again

[image]

rusefillc, 2021-10-23:
@BeerMoneyMotorsports please reproduce with latest snapshot, error message is more detailed now

BeerMoneyMotorsports, 2021-12-14:
i was able to reproduce this recently.  i occurred when the laptop remained connected but closed.  I will attempt to reproduce again and get more detail.

mck1117, 2021-12-14:
Happen to remember what it said? I think we added what system caused the error to the message.

BeerMoneyMotorsports, 2021-12-15:
@mck1117

[image]

rusefillc, 2022-03-21:
@elDominio had it

[image]

rusefillc, 2022-03-21:
@mck1117 can you have another look? at least time for more statements

mck1117, 2022-05-13:
@NMSTEC had it
[image]

mck1117, 2022-05-13:
all of these too far values seem to be close to 32b overflow, which hints at this maybe being an integer overflow somewhere.

log2(1274220776) = 30.2
log2(1395092012) = 30.4

rusefillc, 2022-05-15:
very fresh master on proteus F4
I was playing with iDrive and got distracted unit was just idling connected to rusEFI console

```
2022-05-14_20_47_17_320: EngineState: CRITICAL error: scheduleByTimestampNt() too far: 3999995 scheduleForLater 20220514@proteus
2022-05-14_20_47_17_557: EngineState: CRITICAL error: scheduleByTimestampNt() too far: 3999995 scheduleForLater 20220514@proteus
2022-05-14_20_47_17_758: EngineState: CRITICAL error: scheduleByTimestampNt() too far: 3999995 scheduleForLater 20220514@proteus
```

[image]

rusefillc, 2022-05-15:
[image]

rusefillc, 2022-09-01:
Fixed by #4174

closed 2022-09-01 by rusefillc