rusefi/rusefi #3424: Let's remove "enable boost control"

rusefillc, 2021-10-29:
I believe we do not need explicit "enable boost control" setting

"control mode" field should be enabled as long as we have EITHER solenoid output selected OR one of H-bridge functions is set to Wastegate

We also need an explicit comment on that dialog

We also need to split open loop target from boost control enable since current setup is confusing. I like the was closed loop is done as a separate dialog

rusefillc, 2021-10-29:
Relates to #3385

mck1117, 2021-10-29:
> We also need to split open loop target from boost control enable since current setup is confusing. I like the was closed loop is done as a separate dialog

I don't understand this, can you expand?

rusefillc, 2021-10-29:
this dialog mixed things which should not be mixed - HW and enable on left, with open loop on right

[image]

rusefillc, 2021-10-29:
See also #3429