rusefi/rusefi #3427: proportional spark cut

mmatousek89, 2021-10-29:
lets cut sparks in controlled way , cut exactly needed percent of sparks without risk of overheating one cylinder and fouling plug with unburnt fuel on other cylinder. and on top of that preferably adjust percent of cut from LUA

 ideal is ign+fuel cut ( whole event cut ) but thats dificult to do properly due to fuel pudling in intake and other challenges related to fuel cut and fuel. restore in proper way.... its much easier on gdi...

you can allso RANDOMISE by registering number of cut on every single cylinder and aply cut on cylinder with lowest number of previuous cuts 

bigger question for me is how to aply 30%cut properly, you have to choose some time window to have period of "pwm" controll but in same time you want to distribute cutted events as evenly as possible

when just counting loud:

4cyl 6000rpm
100revs per sec
200events per sec
at least 10hz spark cut pwm freq
20events to calculate PWM
thats 5% resolution of "SparkCut" variable.
or we can calculate over series of 5 pwm periods and do one more or less cut  in one period to obtain more precise resolution.

but we still need to evenly distribute cuts over one period, cuting 10 consequent events and leaving 10consequent events for 50% cut doesnt seem to me like good idea,
it will be beter to cut one, leave one, cut one and so on to obtain smooth engine run even when cutting 50% of sparks

OrchardPerformance, 2021-11-05:
Can you be specific about the operating condition you would want to trigger this controlled cut under?

rusefillc, 2021-11-05:
this is all about launch control

mmatousek89, 2021-11-05:
allso gear cur, soft rev limiter, pit limiter, maybee traction controll and allso antilag and torque correction

rusefillc, 2021-11-16:
#3385

rusefillc, 2021-11-17:
```
setSparkSkipRatio(0.3)

function onTick()
end
```

[image]

[image]

[image]

closed 2021-11-17 by rusefillc