rusefi/rusefi #3433: cache.zip in repo is that a problem or not

rusefillc, 2021-10-30:
we like to re-generate cache.zip somes a few times a day

cache.zip being a binary artifact is probably not good in terms of git practices and repo size

technically all we need is "cache" folder with files in it, not a zip achieve

mck1117, 2021-10-30:
do we need a cache at all? Why not generate every time?

rusefillc, 2021-10-30:
We do not want signature to change without a reason I guess?

mck1117, 2021-10-30:
But the signature is a function of the _input_ files, right?

mck1117, 2021-10-30:
just store a file `last_signature.txt`, and if the current signature doesn't match that, then generate the files

rusefillc, 2021-10-30:
huh? signature is random, it never matches?

mck1117, 2021-10-30:
[image]

SIGNATURE_HASH is based only on the content of the input files, no random

rusefillc, 2021-10-30:
@andreika-git do you remember why do we need cache.zip?

andreika-git, 2021-10-31:
> @andreika-git do you remember why do we need cache.zip?

I don't remember why the file is stored in the repo. Maybe to allow local incremental gen_config calls? When I add something small to the config, I don't want to fully rebuild everything...

mck1117, 2021-10-31:
We don't even use it for incremental rebuilds though, plus it's just not that slow to do a generate.  `./gen_config_default.sh` takes 2 seconds on my machine.

PR #3639 (2021-12-02): remove cache zip, rely on LazyFile instead
Remove the zip caching logic.  This had a few problems:

1. every time we changed config, it commits a new zip file to the repo.  This file is binary, so it causes significant repo bloat.
2. It fails to catch changes to the generation tool that cause a change to the output. edit: there exist a few invalid txt files already in the repo that weren't detected!
3. It doesn't save enough time to be worth it. `time ./gen_config_default.sh` takes 1.9 seconds for me.

fixes #3433
closed 2021-12-02 by rusefillc