rusefi/rusefi #3440: generate kicad diffs task is not well

rusefillc, 2021-11-01:
[image]

to be honest we are not really using it. but it's a cool thing to have.

disable but not remove?
not invest into repairs unless?

chuckwagoncomputing, 2021-11-01:
I've been working on this.
The root cause is that I edited the EEC-V breakout in Kicad 5.99, and the fork of KiCad-Diff that the workflow uses is old and doesn't work with 5.99.
First I tried the current version, but it had a few problems:
https://github.com/Gasman2014/KiCad-Diff/issues/71
@leoheck solved this problem with this branch: https://github.com/leoheck/KiCad-Diff/tree/improve-plotpcb-zoom-on-5.99
The remaining problem is that for some reason the SVGs generated by the new version of KiCad-Diff won't display in Firefox for me. I've tried on my Archlinux laptop, and on the Windows computer I use at work. However on Leo's computer running Ubuntu, Firefox shows them just fine. I'm not sure what's up yet, but not working on Firefox isn't acceptable to me.

rusefillc, 2022-01-24:
Does it help that we had KiCAD 6.00 release @chuckwagoncomputing ?

leoheck, 2022-01-24:
Check the discussion happening here https://github.com/Gasman2014/KiCad-Diff/pull/77

chuckwagoncomputing, 2022-01-25:
I haven't done anything with it because Github's Action runners are on Ubuntu 20.04, which doesn't have Kicad 6.0, but I think we could continue using the nightlies.

chuckwagoncomputing, 2022-02-05:
@leoheck I'm not sure where the proper place to report this is, but I've been trying with the pcm and wxwidges_and_pcm branches and am getting a strange artifact on the layer that I edited to test KiCad-Diff:
[image]
The only thing I changed on the PCB was that I added " test" to a silkscreen text. Only that layer has the artifact.
Here is the PCB source:
https://github.com/rusefi/rusefi/tree/master/hardware/Breakout_104pin_EEC-V-Connector
Here's the script I used:
https://github.com/chuckwagoncomputing/kicad-libraries/blob/ssh-diffs/generate_diffs.sh
And here's the workflow file:
https://github.com/chuckwagoncomputing/rusefi/blob/ssh-diffs/.github/workflows/gen-diffs.yaml
Here's a tar of the output. This was generated with the wxwidges_and_pcm branch.
[diffs.tar.gz](https://github.com/rusefi/rusefi/files/8007275/diffs.tar.gz)

leoheck, 2022-02-14:
@chuckwagoncomputing, I don't know about this project here I just replied here because someone quoted one of my comments in another issue.

Now talking about your issue. It looks like Kicad-Diff is working fine as I could see by myself here. Kicad 6 still has issues plotting svg files. I saw the issue here is in the logos of your board.

Removing those logos I could remove the issue as you can see here in the `local` (left) pane.
[image]

In summary, this is an issue that Kicad still has to fix in the next version of the tool. Unfortunately, I cannot do much.

chuckwagoncomputing, 2022-02-19:
I guess we have the option of merging my branch, and having happy task producing unhappy diffs, and when Kicad gets fixed it will start working by itself.

rusefillc, 2022-02-20:
@chuckwagoncomputing your call. Realistically we are not using kicad diffs much - with Hellen we need "gerber diffs" not "kicad diffs" since Hellen boards are only full in gerber form.

PR #4951 (2023-01-07): Kicad diffs - use ssh, new version of Kidiff
Kidiff generation seems pretty happy now, let's go ahead and fix #3440
closed 2023-01-07 by rusefillc