rusefi/rusefi #3442: Something something Automatic Compression Release

rusefillc, 2021-11-01:
Maybe probably Lua? see https://www.rusefi.com/forum/viewtopic.php?f=4&t=2187

mck1117, 2021-11-01:
Lua.

rusefillc, 2021-11-11:
if got event in last 5 seconds, until we see RPM, engage solenoid

rusefillc, 2021-11-14:
Done

```

outputIndex = 0
startPwm(outputIndex, 100, 0)

rpmLimitSetting = findSetting("compReleaseRpm", 300)
compReleaseDulationLimit = findSetting("compReleaseDur", 6000)

function onTick()
	rpm = getSensor("RPM")
-- handle nil RPM, todo: change firmware to avoid nil RPM
	rpm = (rpm == nil and 0 or rpm)
    print('Rpm ' .. rpm)
	print('getTimeSinceTriggerEventMs ' .. getTimeSinceTriggerEventMs())

	enableCompressionReleaseSolenoid = getTimeSinceTriggerEventMs() < compReleaseDulationLimit and rpm < rpmLimitSetting
    duty = enableCompressionReleaseSolenoid and 100 or 0
    print("Compression release solenoid " .. duty)
	setPwmDuty(outputIndex, duty)
end

```

rusefillc, 2021-11-14:
[image]

[image]

rusefillc, 2021-11-16:
Done

closed 2021-11-16 by rusefillc