rusefi/rusefi #3443: do not control VVT during cranking

rusefillc, 2021-11-01:
during cranking at least on Honda K we get very weird VVT angles, also we do not have oil pressure anyway

I guess we should not bother with VVT PID during cranking?

mck1117, 2021-11-03:
random OEM tune has a minimum VVT operation RPM

[image]

blundar, 2021-11-03:
It is very common in OEM land for there to be a delay between startup and
VCT control.  I think there is a lot of concern about getting oil pressure
to stabilize before diverting oil pressure to do additional things.

mck1117, 2021-11-03:
> I think there is a lot of concern about getting oil pressure to stabilize before diverting oil pressure to do additional things.

also, preventing integrator windup during a period of no oil pressure, then oil pressure arrives, and it bonks the cam in to the stop

mck1117, 2021-11-03:
so I see a need for two things then:
1. minimum VVT control rpm, if below RPM, reset pid controller and output 0 duty
2. minimum VVT engine run time, if engine not running long enough, reset PID and output 0 duty

rusefillc, 2021-11-03:
Yes, but maybe reuse cranking RPM for 1.?

mck1117, 2021-11-03:
Yeah, that's probably fine for now unless we come up with a good reason that doesn't work. #2 mostly covers it "engine just started and is at low RPM" anyway.

commit b7760039 (2022-04-10): do not control VVT during cranking fix #3443
closed 2022-04-10 by rusefillc