rusefi/rusefi #3447: On Proteus Main relay control

rusefillc, 2021-11-02:
We have ``EFI_MAIN_RELAY_CONTROL`` which control is we simply do "1" or vbatt conditional logic for main relay control

most hellens for instance have 
```
DDEFS += -DEFI_MAIN_RELAY_CONTROL=TRUE
```

it looks like Proteus is NOT using this feature and has less smart MR control

is this a feature or a bug?

rusefillc, 2021-11-03:
looks to me like at least current Proteus is good for MR control based on VBatt

[image]

rusefillc, 2021-11-03:
Same on 0.2

[image]

mck1117, 2021-11-03:
yes, I think that would be fine to do.  I seem to remember the EFI_MAIN_RELAY logic was maybe super complicated, but that's a separate issue.

rusefillc, 2021-11-03:
@mck1117 you recall correctly - at least at some point EFI_MAIN_RELAY extremely complicated. I believe it was simplified a few months back and one day we shall re-implement the advanced behaviors in a much cleaner way.

closed 2021-11-03 by rusefillc