rusefi/rusefi #3448: Proteus piggyback ford mustang 2017 v6: water injector

rusefillc, 2021-11-03:
Input:
RPM (parallel with OEM VR crank) (or CAN in the future)
MAP sensor (external sensor NOT parallel but dedicated for rusEFI)
TPS (parallel with OEM)

flow sensor: linear analog calibration TBD
low level in water tank. on/off digital switch.

Outputs:
water pump on/off
solenoid duty cycle
warning light

mck1117, 2021-11-03:
Sounds like lua.

rusefillc, 2021-11-03:
Argenis parallel with OEM VR or Hall sensor is the biggest risk I see here

*first step to test would be parallel OEM RPM sensing*

second priory all other OEM sensor I am less worries about those

rusefillc, 2021-11-03:
Argenis gets the car on November 12th

mck1117, 2021-11-03:
just sniff an ignition output to measure RPM if we don't care about phasing

mck1117, 2021-11-03:
Or just sniff all this stuff over CAN.

OrchardPerformance, 2021-12-05:
For simple water meth control we have been able to just PWM the pump via a GPIO map.