rusefi/rusefi #3451: voltage compensate solenoids

mck1117, 2021-11-03:
We can voltage compensate various solenoids to correct for battery voltage effects.  Provides more predictable behavior during cranking, failed alternator, etc.

- idle solenoid
- VVT solenoids

rusefillc, 2021-11-03:
Since you mention VVT see #3443

mck1117, 2021-11-03:
example working PWM supply voltage compensation from https://github.com/mck1117/wideband:

```
        float heaterVoltage = GetVoltageForState(state, heaterEsr);

        // Limit to 11 volts
        if (heaterVoltage > 11) {
            heaterVoltage = 11;
        }

        // duty = (V_eff / V_batt) ^ 2
        float voltageRatio = heaterVoltage / batteryVoltage;
        float duty = voltageRatio * voltageRatio;

        if (batteryVoltage < 23)
        {
            // Pipe the output to the heater driver
            heaterPwm.SetDuty(duty);
        }
        else
        {
            // Overvoltage protection - sensor not rated for PWM above 24v
            heaterPwm.SetDuty(0);
        }
```