rusefi/rusefi #3457: Limit ETB to < 100%

duhafnusa4, 2021-11-04:
@mck1117 and I through testing on my car found problems with ETB when value is 100% or greater. 

We need to limit the ETB to less than 100%. @mck1117 feel free to comment

rusefillc, 2021-11-04:
@mck1117 can you please elaborate? is this about TS UI limitation? is this about defaults? is this about software changes?

OrchardPerformance, 2021-11-05:
This is to prevent the ETB driving past 90 degrees, this is possibly a software change to allow a user settable limit to prevent this situation. 

See image below with the ETB at 100% but sitting about 5-10 degrees past fully open (90deg). 
[image]

rusefillc, 2021-11-05:
So is this a chance for auto-calibration to calibrate 100% point as NOT the full open but magic percentage of full open?

duhafnusa4, 2021-11-05:
I recognize that thumb, and the photo i took doesnt do it justic, Id say its closer to 10% past 100%

PR #3469 (2021-11-05): configurable etb maximum
fix #3457
closed 2021-11-05 by rusefillc