rusefi/rusefi #3461: Nissan MR18DE trigger

mck1117, 2021-11-04:
Do we support this trigger pattern?

[image]

rusefillc, 2021-11-04:
Does not looks like we support https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#nissan :(

What's the evil plan?

2004–2012 Nissan Tiida/Versa

mck1117, 2021-11-04:
I want a VVT test mule

2004-2012 Nissan Tiida/Versa has VVT

There is a 2004-2012 Nissan Tiida/Versa in my driveway 😈

rusefillc, 2021-11-04:
@mck1117 https://github.com/rusefi/kicad-libraries/commit/eb70637ed6f87a6ff3a6fb60054b460edcd2e4e6 see https://rusefi.com/forum/viewtopic.php?f=4&t=2195

mck1117, 2021-11-04:
it looks like a close relative of the VQ cam pattern but 1-3-4-2 instead of 1-4-5-2-3-6

[image]

mck1117, 2021-11-04:
crank on MR18DE is 16 teeth, then two teeth of skip, aka it's 36-2-2

mck1117, 2021-11-05:
we can close for now, will revisit when I can log exact trigger from the car.

closed 2021-11-05 by mck1117