rusefi/rusefi #3463: + symbol required to make TS tips show

OrchardPerformance, 2021-11-05:
[image]

Image above shows a case where there is a TS tip but it is not displaying due to a lack of +

I have not checked _every_ case but it looks like it is true for all the tips without + symbols at the start.

OrchardPerformance, 2021-11-05:
Shows up as lines 382 and 383 in VS code 
[image]

rusefillc, 2021-11-05:
@OrchardPerformance weird. Unless we somehow speak about different fields? See image

[image]

rusefillc, 2021-11-09:
@OrchardPerformance ?

OrchardPerformance, 2021-11-10:
It looks like I picked a terrible example, actually possibly the worst available. 
I pulled that out while busy at work, agree there is a double comment there, it was one to correct from the PR. 

Checking better see below:
[image]
[image]
[image]

[image]
[image]
[image]

rusefillc, 2021-11-10:
I've pulled latest .ini and I seem to have both tooltips?

[image]

[image]

OrchardPerformance, 2021-11-10:
OK that is weird. 
It was a late October bundle I took those screen shots of, will check the exact date but I think it's the perogi day release. 

Let me check a couple of releases and update in an hour or so.

rusefillc, 2021-11-10:
I do not follow. Why do we look at releases at all? I believe we only care about how SNAPSHOT works.

I am failing to convince you that I've changed this in SNAPSHOT :( Any release is simply irrelevant here is not it?

I can get you exact date of when this was changed but my concern stands: we only care about snapshot nothing else.

OrchardPerformance, 2021-11-10:
So when I asked: 
[image]
The correct answer was "this has been changed in the last few weeks so it is no longer needed" 

8 days ago when the PR was submitted and neither Matt nor myself knew why + symbols were missing the key information was really "This was changed recently enough that the current release is obsolete and the workflow has changed in the last couple of weeks" 

When I raised this ticket wondering why neither I not USERS could see these tips the vital information was "I have changed this since the release" 

It is not about convincing me of anything, simply letting me know that a change has happened so recently that it is only in snapshot and I might see different results if I am not using the snapshot. 

I am referring to the release as I need to be able to see what users see and why they are asking questions that could be answered in tool tips. 
Obviously when there is no tip in the release and I see one with a missing + in the current git version I am going to think there is a problem with missing + because no one said things changed since the version the users are looking at.

closed 2021-11-10 by OrchardPerformance

rusefillc, 2021-11-10:
in my opinion releases are not relevant for development and this was all about development

knowing exact user version is helpful for troubleshooting - some users use some releases some users use some snapshots for reasons. "latest release" does not hold a very special place in my opinion.

overall all my life is in snapshot. latest release was a month ago, that's about 8 times longer than my attention span.

Note how I was pointing at "master" which is maybe where I should have be more clear since "master" is git specific talk to "snapshot"

[image]

OrchardPerformance, 2021-11-11:
This is why we need some stability