rusefi/rusefi #3465: leverage #3464 for existing tables

rusefillc, 2021-11-05:
@racer-coder can you please leverage #3464 for some of the current tables like
```
custom map_estimate_table_t	2*@@FUEL_RPM_COUNT@@x@@FUEL_LOAD_COUNT@@ array, U16, @OFFSET@, [@@FUEL_RPM_COUNT@@x@@FUEL_LOAD_COUNT@@], "kPa", {1/@@PACK_MULT_MAP_ESTIMATE@@}, 0, 0, 100, 2
custom ve_table_t 4*@@FUEL_RPM_COUNT@@x@@FUEL_LOAD_COUNT@@ array, F32, @OFFSET@, [@@FUEL_RPM_COUNT@@x@@FUEL_LOAD_COUNT@@], "%", 1, 0, 0, 999, 2
custom lambda_table_t @@FUEL_RPM_COUNT@@x@@FUEL_LOAD_COUNT@@ array, U08, @OFFSET@, [@@FUEL_RPM_COUNT@@x@@FUEL_LOAD_COUNT@@], "deg", {1/@@PACK_MULT_LAMBDA_CFG@@}, 0, 0.6, 1.5, 2
custom afr_table_t @@FUEL_RPM_COUNT@@x@@FUEL_LOAD_COUNT@@ array, U08, @OFFSET@, [@@FUEL_RPM_COUNT@@x@@FUEL_LOAD_COUNT@@], "deg", {1/@@PACK_MULT_AFR_CFG@@}, 0, 0, 25, 1

```

racer-coder, 2021-11-05:
I have a couple of other changes in mind first.

racer-coder, 2021-11-05:
Huh I can't seem to assign this to myself even though I'm a member of rusefi?

rusefillc, 2021-11-05:
being a member allows at least me to assign it now :)

PR #3473 (2021-11-06): Convert existing tables to use the simpler 3d table definition style in rusefi_config.txt
This fixes #3465.
commit 16378dcf (2021-11-06): Convert existing tables to use the simpler 3d table definition style in rusefi_config.txt (#3473)

This fixes #3465.
closed 2021-11-06 by rusefillc