rusefi/rusefi #3481: TS project is not happy

rusefillc, 2021-11-06:
[image]

@racer-coder to be honest I do not know what is wrong exactly but it's next to hpfp changes from yday

racer-coder, 2021-11-06:
It's a capitalization bug, but I'm trying to figure out why I can't reproduce it

rusefillc, 2021-11-06:
i had old project
I went to update .ini for existing project and got bug

capitalization between what and what? I am blind on top of everything

closed 2021-11-06 by rusefillc

racer-coder, 2021-11-06:
How did you get that window to pop up?  I had to dig into TunerStudioAppDebug.txt to find that message.

rusefillc, 2021-11-06:
[image]

PR #3482 (2021-11-06): Fix capitalization of subMenu for TS
This fixes #3481
commit f3df36b4 (2021-11-06): Fix capitalization of subMenu for TS (#3482)

This fixes #3481