rusefi/rusefi #3483: openblt doesn't work proteus f7

mck1117, 2021-11-06:
Current proteus f7 images don't work, no LEDs, no nothing.  Home compiled without openblt works fine.

dron0gus, 2021-11-06:
That was expected.
Can you check that settings in proteus/openblt/main.c looks correct for your board?
Can you confirm that external oscillator is 8MHz on Proteus?
Currently STM32F767 is selected for Proteus. Is it correct?
Does your board need additional setup? Currently code setups clocks, gpio, uart and can1. And one gpio for LED - GPIOE3

dron0gus, 2021-11-06:
BTW, can you confirm that you are flashing `rusefi_openblt.dfu`? Or `openblt_proteus.dfu` and then `rusefi.dfu`

mck1117, 2021-11-06:
> That was expected.

Maybe let's not break things on purpose 😛 

It is an F767. There is no guarantee about oscillator speed....can we use HSI instead since it's always known?

dron0gus, 2021-11-06:
My fault. I did not expect such rush merge. Plan was to merge everything but not 55621988ef02c7a3221ad2f3f58bcf8fb46bed77

rusefillc, 2021-11-06:
@dron0gus do we need a separate ticket to auto-detect OSC in OpenBLT?

dron0gus, 2021-11-06:
@rusefillc only if we have any board that may be assembled with random OSC. Do we have such?

rusefillc, 2021-11-06:
@dron0gus we absolutely LOVE auto-OSC in rusEFI
one of hellen boards we have 8. 16 and 25 units.

dron0gus, 2021-11-08:
Correct images should be included in bundle after https://github.com/rusefi/rusefi/pull/3486

rusefillc, 2021-11-08:
@dron0gus what about dynamic vs hard-coded osc speed?

rusefillc, 2022-01-16:
poke @dron0gus 

@NMSTEC FYI

dron0gus, 2022-01-16:
Someone with access to Proteus HW should test this https://github.com/rusefi/rusefi/pull/3797 .

rusefillc, 2022-01-17:
I've merged #3797 to see if HW CI would pass and because that's the simplest way for me to test

it has failed

on the bright side I had a chance to make progress https://github.com/rusefi/rusefi/commit/0f5b2366ed286d5a51d7ace3741e8949f2578918

I've reverted for now https://github.com/rusefi/rusefi/commit/1ac9d72ed52439fff97bf03538b91c728673204d

@dron0gus let's make an MRE attempt since you have MRE?

rusefillc, 2022-02-12:
making progress with test

```
dfu-util -a 0 -D rusefi.dfu -R
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 0483
Match product ID from file: df11
dfu-util: Patching string descriptor 4 length (was 166, received 112)
Multiple alternate interfaces for DfuSe file
Opening DFU capable USB device...
Device ID 0483:df11
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
File contains 1 DFU images
Parsing DFU image 1
Target name: EncedoKey
Image for alternate setting 0, (2 elements, total size = 510328)
Setting Alternate Interface #0 ...
Parsing element 1, address = 0x08000000, size = 12016
Erase           [=========================] 100%        12016 bytes
Erase    done.
Download        [=========================] 100%        12016 bytes
Download done.
Parsing element 2, address = 0x08008000, size = 498296
Erase           [=========================] 100%       498296 bytes
Erase    done.
Download        [=========================] 100%       498296 bytes
Download done.
Done parsing DfuSe file
dfu-util: can't detach
Resetting USB to switch back to Run-Time mode

```

rusefillc, 2022-02-12:
```
firmware/ext/openblt/Host/Source/LibOpenBLT/build

vi ../CMakeLists.txt

grep VERSION ../CMakeLists.txt
cmake_minimum_required(VERSION 3.10)

cmake ..
make
cd ../..
cd BootCommander/build
vi ../CMakeLists.txt
cmake ..
make

```

dron0gus, 2022-02-20:
Test with CANable Z PRO and MRE 0.5.0.
```
$ dmesg | tail
[212471.989916] usb 2-1.8.4.1.2: new full-speed USB device number 52 using ehci-pci
[212472.208871] usb 2-1.8.4.1.2: New USB device found, idVendor=1d50, idProduct=606f
[212472.208878] usb 2-1.8.4.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[212472.208880] usb 2-1.8.4.1.2: Product: candleLight USB to CAN adapter
[212472.208883] usb 2-1.8.4.1.2: Manufacturer: bytewerk
[212472.208884] usb 2-1.8.4.1.2: SerialNumber: 00240020564D510A20353632
$ ifconfig can0
can0: flags=128<NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 10  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Don't forget to enable 120R load on CAN bus. `TRES` switch on CANable adapter.

1. Enable OpenBLT for mre with F4:
```
$ git diff
diff --git a/firmware/config/boards/microrusefi/compile_mre_f4.sh b/firmware/config/boards/microrusefi/compile_mre_f4.sh
index 339abdb196..fa20445231 100644
--- a/firmware/config/boards/microrusefi/compile_mre_f4.sh
+++ b/firmware/config/boards/microrusefi/compile_mre_f4.sh
@@ -4,6 +4,6 @@ export PROJECT_BOARD=microrusefi
 export PROJECT_CPU=ARCH_STM32F4
 export EXTRA_PARAMS="-DSHORT_BOARD_NAME=mre_f4"
 
-# export USE_OPENBLT=yes
+export USE_OPENBLT=yes
 
 bash ../common_make.sh
```
2. Rebuilding. Don't forget to ``make clean`` after changing ``USE_OPENBLT`` option!
```
$ make clean
$ ( cd ./config/boards/microrusefi/ ; time CCACHE_DIR=../../../cc_cache bash ./compile_mre_f4.sh )
...
common_make.sh: invoking hex2dfu for incremental rusEFI image
Element 0: adding OpenBLT-style crc @ 0x0000001c: 0xaff320a2
Total elements: 1
 Element 0:
  Data Start:  0x08008000
  Data Length: 588520 bytes
Done.
common_make.sh: invoking hex2dfu for OpenBLT
Total elements: 1
 Element 0:
  Data Start:  0x08000000
  Data Length: 12036 bytes
Done.
common_make.sh: invoking hex2dfu for composite rusEFI+OpenBLT image
Element 1: adding OpenBLT-style crc @ 0x0000001c: 0xaff320a2
Bin file: base: 0x08000000s, len 621288 bytes
Total elements: 2
 Element 0:
  Data Start:  0x08000000
  Data Length: 12036 bytes
 Element 1:
  Data Start:  0x08008000
  Data Length: 588520 bytes
Done.
common_make.sh: build folder content:
total 32548
drwxrwxr-x 2 dron dron     4096 фев 20 13:29 lst
drwxrwxr-x 2 dron dron    20480 фев 20 13:29 obj
-rwxrwxr-x 1 dron dron   588520 фев 20 13:29 rusefi.bin
[... 197 lines cut ...]
  -> Seed/Key file: None
  -> Connection mode: 0
Detected transport layer: XCP on CAN
Using transport layer settings:
  -> Device: can0 (channel 0)
  -> Baudrate: 500000 bit/sec
  -> Transmit CAN identifier: 667h
  -> Receive CAN identifier: 7E1h
  -> Use 29-bit CAN identifiers: No
Loading firmware data from file...[OK]
  -> Number of segments: 2
  -> Base memory address: 0x08008000
  -> Total data size: 588512 bytes
Connecting to target bootloader...[TIMEOUT]
Attempting backdoor entry (reset system if this takes too long)...
````
4. Power on MRE or reset it.
```
Attempting backdoor entry (reset system if this takes too long)...[OK]
Erasing 488 bytes starting at 0x08008000...[OK]  
Erasing 588024 bytes starting at 0x080081f0...[OK]  
Programming 488 bytes starting at 0x08008000...[OK]  
Programming 588024 bytes starting at 0x080081f0...[OK]  
Finishing programming session...[OK]
````

rusefillc, 2022-02-20:
just pushed https://github.com/rusefi/rusefi/commit/e2101d13f9b5c99b929e4b2f32b75d0cdd6919a6 in light of #3802

rusefillc, 2022-02-21:
Worked for me!
```
debian@debian:~/rusefi/firmware$ ext/openblt/Host/BootCommander -t=xcp_can -d=can0 deliver/rusefi_update.srec
--------------------------------------------------------------------------
BootCommander version 1.04.02. Performs firmware updates on a micro-
controller based system that runs the OpenBLT bootloader.

Copyright (c) 2017 by Feaser  http://www.feaser.com
-------------------------------------------------------------------------
Processing command line parameters...[OK]
Detected firmware file: deliver/rusefi_update.srec
Detected session protocol: XCP version 1.0
Using session protocol settings:
  -> Timeout T1: 1000 ms
  -> Timeout T3: 2000 ms
  -> Timeout T4: 10000 ms
  -> Timeout T5: 1000 ms
  -> Timeout T6: 50 ms
  -> Timeout T7: 2000 ms
  -> Seed/Key file: None
  -> Connection mode: 0
Detected transport layer: XCP on CAN
Using transport layer settings:
  -> Device: can0 (channel 0)
  -> Baudrate: 500000 bit/sec
  -> Transmit CAN identifier: 667h
  -> Receive CAN identifier: 7E1h
  -> Use 29-bit CAN identifiers: No
Loading firmware data from file...[OK]
  -> Number of segments: 2
  -> Base memory address: 0x08008000
  -> Total data size: 500828 bytes
Connecting to target bootloader...[OK]
Erasing 488 bytes starting at 0x08008000...[OK]
Erasing 500340 bytes starting at 0x080081f0...[OK]
Programming 488 bytes starting at 0x08008000...[OK]
Programming 500340 bytes starting at 0x080081f0...[OK]
Finishing programming session...[OK]
```

rusefillc, 2022-02-21:
i have MRE which sporadically stays in CAN bootloader

reverting openblt in MRE for now https://github.com/rusefi/rusefi/commit/b1127349ad8736686c44061dc1bdaa0f58f04427

rusefillc, 2022-05-10:
https://github.com/rusefi/rusefi/commit/57840b19982d807cb3bb1cbfba0d1de8336219aa is the fix