rusefi/rusefi #3485: unit tests build time: Address sanitizer

rusefillc, 2021-11-06:
Can we please NOT run Address sanitizer in "default' makefile unit_tests build but only on github actions?

I am looking to get myself a quicker local build :)

rusefillc, 2021-11-06:
@mck1117 can you help?

@racer-coder FYI

racer-coder, 2021-11-06:
Normally i'd recommend clang for faster compile times but it was even slower than gcc.  It might be related to coverage compilation too.

mck1117, 2021-11-06:
Code coverage is much more expensive than asan. I tested it and asan made almost no difference.

racer-coder, 2021-11-06:
I also got a big improvement removing -Wa alms blah blah

mck1117, 2021-11-07:
> I also got a big improvement removing -Wa alms blah blah

good call, that saves ~20% for me

mck1117, 2021-11-07:
Also, you do really want asan enabled for development.  That's when you're going to be making leaks/overruns, so you want it always there.

PR #3489 (2021-11-07): unit test build times
- don't emit assembly listings for unit tests (nobody looks at these?), worth about 20%
- disable code coverage unless specified, and specify it in CI runs, worth another 15%

Total improvement: 32% reduction in compile time.

Address sanitizer stays on, since it was only worth about 4% on my machine.

fix #3485
closed 2021-11-07 by rusefillc