rusefi/rusefi #3488: Remove NB1 trigger

rusefillc, 2021-11-07:
NB1 trigger is same as NB2 VVT configuration

in order to simplify life we shall remove NB1 trigger and use NB2 on NB1

mck1117, 2021-11-07:
@BeerMoneyMotorsports presumably you could test this at some point, set an NB1 board to NB2 vvt trigger settings, and see what happens?

BeerMoneyMotorsports, 2021-11-07:
Yes. Nb vvt should be renamed to NB amd other nb deleted

rusefillc, 2022-01-31:
@BeerMoneyMotorsports this has happened!

closed 2022-01-31 by abelom

rusefillc, 2022-02-01:
@BeerMoneyMotorsports 

@BeerMoneyMotorsports NB1 trigger is gone, people with old NB1 trigger will get fatal error

10:18
on hellen NB1 boards we technically can make stuff hard-coded (edited) 
10:18
so that new firmware would change it's own settings
10:19
nope we cannot :disappointed:
10:20
if we go 'board override' it would be impossible to go 36/1
10:20
i assume you want 36/2 as option
10:20
so all we can 'override' is VVT
10:20
and trigger pins
10:20
@BeerMoneyMotorsports let me know

commit 1d64929f (2022-01-31): Remove Miata MX5 NB1 trigger fix #3488