rusefi/rusefi #3498: Bells & whistles: board-specific image into console splash screen

rusefillc, 2021-11-07:
at the moment I have to carefully read title text to make sure I've launched the right console for all the HW I have

how can we differentiate splash screens a bit more? custom shades of background color? a small image of proteus VS MRE vs hellen? else?

[image]

rusefillc, 2021-11-07:
@ksmola can I interest you in this one?

duhafnusa4, 2021-11-15:
[image]

duhafnusa4, 2021-11-15:
[image]

rusefillc, 2021-11-16:
@duhafnusa4 love it! will try placing those somewhere on the splash screen

commit f8abb380 (2022-02-01): Bells & whistles: board-specific image into console splash screen fix #3498

done for now