rusefi/rusefi #3503: Consider dropping ChibiOS CH_DBG_SYSTEM_STATE_CHECK

racer-coder, 2021-11-08:
It adds a ton of overhead to CriticalSectionLocker.  Something that should be very simple becomes very complex..

Without:
 804b9e4:       f3ef 8311       mrs     r3, BASEPRI
 804b9e8:       b923            cbnz    r3, 804b9f4 <getTimeNowNt+0x14>
 804b9ea:       f3ef 8205       mrs     r2, IPSR
 804b9ee:       2220            movs    r2, #32
 804b9f0:       f382 8811       msr     BASEPRI, r2
(skipping unrelated instructions)
 804b9f8:       9303            str     r3, [sp, #12]

Also note the instruction at 804b9ea becomes completely unnecessary as there is no difference between port_lock and port_lock_from_isr on ARM.  We can optimize that out of ChibiOS if we don't use system state debugging.

With debugging:
 804c844:       f3ef 8311       mrs     r3, BASEPRI
 804c848:       b9d3            cbnz    r3, 804c880 <getTimeNowNt+0x40>
 804c84a:       f3ef 8205       mrs     r2, IPSR
 804c84e:       f3c2 0208       ubfx    r2, r2, #0, #9
 804c852:       b15a            cbz     r2, 804c86c <getTimeNowNt+0x2c>
/* ISR VERSION */
 804c854:       2220            movs    r2, #32
 804c856:       f382 8811       msr     BASEPRI, r2
 804c85a:       4a16            ldr     r2, [pc, #88]   ; (804c8b4 <getTimeNowNt+0x74>)
 804c85c:       6b11            ldr     r1, [r2, #48]   ; 0x30
 804c85e:       2900            cmp     r1, #0
 804c860:       dd01            ble.n   804c866 <getTimeNowNt+0x26>
 804c862:       6b51            ldr     r1, [r2, #52]   ; 0x34
 804c864:       b151            cbz     r1, 804c87c <getTimeNowNt+0x3c>
 804c866:       4814            ldr     r0, [pc, #80]   ; (804c8b8 <getTimeNowNt+0x78>)
 804c868:       f7bb f9da       bl      8007c20 <chSysHalt>
/* non-ISR VERSION */
 804c86c:       2220            movs    r2, #32
 804c86e:       f382 8811       msr     BASEPRI, r2
 804c872:       4a10            ldr     r2, [pc, #64]   ; (804c8b4 <getTimeNowNt+0x74>)
 804c874:       e9d2 010c       ldrd    r0, r1, [r2, #48]       ; 0x30
 804c878:       4301            orrs    r1, r0
 804c87a:       d117            bne.n   804c8ac <getTimeNowNt+0x6c>
 804c87c:       2101            movs    r1, #1
 804c87e:       6351            str     r1, [r2, #52]   ; 0x34
(skipping unrelated instructions)
 804c884:       9303            str     r3, [sp, #12]

rusefillc, 2021-11-08:
in my opinion a better solution would be to have more fine-grained control over this, to remove it ONLY from CriticalSectionLocker maybe in our fork of ChibiOS?

or go to chibios forum and suggest spliting CH_DBG_SYSTEM_STATE_CHECK into two separate things:

CH_DBG_SYSTEM_STATE_CHECK (for everything not critical CriticalSectionLocker)

and CH_DBG_SYSTEM_STATE_CRITICAL_CHECK  for more time-sensitive stuff

mck1117, 2021-11-08:
That state check is certainly actually useful in debug builds, avoiding developer mistakes with lock/unlock behavior.