rusefi/rusefi #3508: trigger shape and PWM shape does not need full byte to represent 1 bit

rusefillc, 2021-11-09:
SingleChannelStateSequence pinStates is taking 8 times more RAM than it should

```
/**
 * int8_t is probably less efficient then int32_t but we need
 * to reduce memory footprint
 *
 * todo: migrate to bit-array to save memory?
 * this would cost some CPU cycles. see std::vector<bool>
 */
typedef trigger_value_e pin_state_t;
```

rusefillc, 2021-11-09:
@racer-coder FYI

mck1117, 2021-11-09:
do we actually need to store state? Isn't this array always going to contain 0101010101010101? can't we compute state by `index & 1`?

racer-coder, 2021-11-09:
Yeah I started down that one but wasn't sure if setState needed thread safety or not.

racer-coder, 2021-11-09:
hahahaha +1 on xor

mck1117, 2021-11-09:
we can add an assert and see if `state != (index & 1)` is ever actually possible, and if so, get rid of those cases so we can avoid storing the state all together!

rusefillc, 2021-11-09:
@mck1117 no it would not contain 0101010101010101

in case of trigger you have two of these working together

0001010000010
0100000101001

mck1117, 2021-11-09:
related:

```
/*
 * todo: should we simply re-use instances used by trigger_decoder?
 * todo: since we are emulating same shape we are decoding
 */
static pin_state_t pinStates1[PWM_PHASE_MAX_COUNT];
static pin_state_t pinStates2[PWM_PHASE_MAX_COUNT];
static pin_state_t pinStates3[PWM_PHASE_MAX_COUNT];
```

racer-coder, 2021-11-09:
I actually think it isn't the bits as bytes but the float that's the problem.  Seems that many cases of SingleChannelStateChange are fed into MultiChannel.  So
1) turning the float into a uint16_t where 0 maps to 1./65536 and 65535 maps to 65536/65536; that's approx 0.011 degrees resolution on a 720 degree cycle, i.e. better than our microsecond timer
2) put all the channels into one byte

yeah 2) wastes some memory but it keeps thing simpler.

I'm working through some of the compilation issues, maybe something will pop up that will negate this plan

mck1117, 2021-11-09:
That's a good start!