rusefi/rusefi #3530: Mercedes "3 Segment" Crank Trigger Support

duhafnusa4, 2021-11-11:
My W124 M111 car has a 3 segment flywheel (one of the segments has a magnet) 

I need a trigger pattern that works with this flywheel. I have gotten the car to start using some BS custom  (2 tooth 1 missing).

This is what the flywheel looks like

[image]

This a trigger log with spark plugs removed, fuel and ign turned off. 

[2021-11-11_11.45.36.csv](https://github.com/rusefi/rusefi/files/7523188/2021-11-11_11.45.36.csv)

*this trigger pattern is used on lots of early/mid 90s mercedes cars(M102, M103, M104, M111, M117, M119) . supporting this is good for rusEFI

duhafnusa4, 2021-11-11:
@mck1117

mck1117, 2021-11-11:
I don't think yours has 3 teeth.  I think it has 2 teeth.

duhafnusa4, 2021-11-11:
I think youre right. ticket updated

duhafnusa4, 2022-02-16:
I think we do for other vehicles. The fuelpump is dead so I won't know anything till I get that in. I'm going to test both triggers on the car for the sake of it