rusefi/rusefi #3532: more specific lua error handling

rusefillc, 2021-11-11:
can we know location closer to error instead of just saying "this script is broken somewhere"?

```
2021-11-11_17_22_59_408: EngineState: LUA error [string "startPwm(0
2021-11-11_17_22_59_516: EngineState: LUA error [string "startPwm(0
2021-11-11_17_22_59_625: EngineState: LUA error [string "startPwm(0
2021-11-11_17_22_59_737: EngineState: LUA error [string "startPwm(0
2021-11-11_17_22_59_846: EngineState: LUA error [string "startPwm(0
```

```
startPwm(0, 800, 0.3)
-- direction
startPwm(1, 80, 1.0)
-- disable
startPwm(2, 80, 0.0)

pid = Pid.new()
pid:setP(2)

function onTick()
  targetVoltage = getAuxAnalog(0)
  
  target = interpolate(1, 0, 3.5, 100, targetVoltage)
-- clamp 0 to 100
  target = math.max(0, target)
  target = math.min(100, target)

  pid:setTarget(target)
  duty = pid:get(analog1)
  
  print('Target ' .. targetVoltage .. ' ' .. target)
  
  
end
```

mck1117, 2021-11-11:
you're trying to print a comma, because that's the delimeter between log messages

rusefillc, 2021-11-11:
@mck1117 you comment is legit but my concern is different/wider?

mck1117, 2021-11-11:
I mean that it clipped off 90% of the error message, there is a more useful message that it didn't fully print

rusefillc, 2021-11-11:
oh so your proposal is to address clipping and revisit

mck1117, 2021-11-11:
to unblock you for now:

`0LUA ERROR loading script: [string \"(startPwm(0, 800, 0.3)...\"]:3: ')' expected (to close '(' at line 1) near 'startPwm'`

rusefillc, 2021-11-11:
not blocked, know where errors are just suggesting we improve one day

PR #3533 (2021-11-19): delimeter is not comma
We can't print commas with `efiPrintf` which is sort of insane.  `$` is less often used, so maybe try that instead?

fixes #3532
closed 2021-11-19 by rusefillc