rusefi/rusefi #3534: Bench Test IAC Valve fails

stefanst, 2021-11-12:
Using this tune: https://rusefi.com/online/view.php?msq=724
Bench testing IAC valve (both valves) fails with the following error:
[image]

stefanst, 2021-11-12:
NM. Settings error

stefanst, 2021-11-12:
This is weird. I played with the idle settings. The error didn't occur. Now it re-occurs repeatedly and reliably. 
New tune here:
https://rusefi.com/online/view.php?msq=725

rusefillc, 2021-11-12:
@stefanst I can confirm that idle bench via TS was pretty broken

FYI secondary idle is definitely not working yet even if we have button in UI

I've made an attempt to fix TS idle bench

stefanst, 2021-11-12:
On the plus side, the outputs behave as expected in open loop mode.

mi-hol, 2023-01-13:
@stefanst I'm looking for a working example of IACV and found your tune.
Is this a correctly working setup?
[image]