rusefi/rusefi #3537: Improve Vehicle Speed Sensor Tool Tips

droneperson42, 2021-11-12:
Under Sensors > Speed Sensor, the tool tips for "revolution to speed mult(coef)" and "Filter Reciprocal (Hz)" aren't that useful in helping users understand what they do and what values should be put in to make it work. I'm not sure exactly what they do but could be along the lines of:

"revolution to speed mult(coef)": Multiplier for converting wheel speed (Hz/rotations per second) to Km/h. Calculated by measuring wheel radius and substituting into this formula .... (Hz > Km/h: [2*pi*wheel_radius*3.6], or of it/s Km/h>Hz: 1/[2*pi*wheel_radius*3.6])

"Filter Reciprocal (Hz)": Number of teeth on (ABS ring/gearbox speed sensor, I'm not sure). Calculates wheel cycles/s or Hz based on how many counts it should expect to read per revolution.

thirstyone, 2021-11-13:
not sure if that's at all possible, but it would be nice to click a button in UI, move a vehicle 100 meters forward, click a button once again and have those coefficients calculated.

BeerMoneyMotorsports, 2021-11-15:
@mck1117 you have another issue for this?

mck1117, 2021-11-15:
this = ?

BeerMoneyMotorsports, 2021-11-15:
Fixing the VSS configuration parameters to be simpler and easier to comprehend.  Not the tool tips specifically.  There was Slack chatter about it which you were involved with.  You were changing to something like entering wheel diameter, gear ratio, etc and having math do it in the background?

rusefillc, 2021-11-15:
@BeerMoneyMotorsports yes

[image]