rusefi/rusefi #3539: maybe lua freezes proteus h7 after 7 seconds of run

rusefillc, 2021-11-12:
proteus h7
power cycle
happy blue LED blinking
MSD opens etc all fun

4-7 seconds later blue stops blinking

i was wring an amazing Lua script while that happened

.hex read attached
[proteus-h7-blue-freeze.zip](https://github.com/rusefi/rusefi/files/7525039/proteus-h7-blue-freeze.zip)

rusefillc, 2021-11-12:
Steps to reproduce:

naked H7, nothing connected via ampseal, USB power
erase
``STM32_Programmer_CLI.exe -c port=usb1 -e all``

rusefillc, 2021-11-12:
simpler version works
```

startPwm(0, 800, 0.1)
-- direction
startPwm(1, 80, 1.0)
-- disable
startPwm(2, 80, 0.0)

pid = Pid.new()
pid:setP(2)
pid:setMinValue(-100)
pid:setMaxValue(100)

biasCurveIndex = findCurveIndex("bias")

function onTick()
  local targetVoltage = getAuxAnalog(0)
  
  local target = interpolate(1, 0, 3.5, 100, targetVoltage)
-- clamp 0 to 100
  target = math.max(0, target)
  target = math.min(100, target)

  print('Decoded target: ' .. target)

  local tps = getSensor("TPS1")
  tps = (tps == nil and 'invalid TPS' or tps)
  print('Tps ' .. tps)

  pid:setTarget(target)
  local output = pid:get(tps)

  local bias = curve(biasCurveIndex, target)
  print('bias ' .. bias)

  print('pid output ' .. output)
  print('')
  
end
				
```

rusefillc, 2021-11-12:
well it's ridiculous

i need to add
```
controlIndex = 0
directionIndex = 1

```

on top to break it

rusefillc, 2021-11-13:
does not reproduce in master :( have we dumped the problem under the rug or do we know that it was fixed?

mck1117, 2021-11-13:
I'm debugging it on the commit before updating luaaa (which seemed to fix it?) to make sure we undertsand what happened.

rusefillc, 2021-11-16:
@mck1117 did you find anything out or got distracted?

mck1117, 2021-11-29:
I couldn't repro it, and nobody has seen it since.

closed 2021-11-29 by mck1117