rusefi/rusefi #3544: MAP phase sensing

rusefillc, 2021-11-14:
Harley does not have a physical CAM sensor.

Harley is asking for MAP sensor to be used for engine phase detection

rusefillc, 2021-11-15:
Let's see if we can grab a nice MAP chart using TS

1) use TunerStudio BETA for better performance
2) set data rate to MAXIMUM
3) change logging configuration to only log MAP ton increase data flow
4) set widest MAP sampling windows so that all angles are sampled

TODO do we have un-averaged fast 'raw' decoded MAP or MAP voltage @mck1117 ?

NMSTEC, 2021-11-24:
We've attempted to take a high speed MAP debug log with TS Beta but MAP values do not refresh often, Andrey wonders if code changes would be needed

@mck1117 can you please look at 
[2021-11-24_11.01.17.mlg.zip](https://github.com/rusefi/rusefi/files/7598143/2021-11-24_11.01.17.mlg.zip)

NMSTEC, 2021-11-24:
[image]

NMSTEC, 2021-11-24:
[DataLogs.zip](https://github.com/rusefi/rusefi/files/7598949/DataLogs.zip)

NMSTEC, 2021-11-24:
[image]

NMSTEC, 2021-11-24:
[map w enginephase.zip](https://github.com/rusefi/rusefi/files/7598964/map.w.enginephase.zip)

rusefillc, 2021-11-24:
@mck1117 I do not trust these flat regions those are too constant. Does this look suspicious to you?

NMSTEC, 2021-11-25:
[image]
[2021-11-24_16.03.29.mlg.zip](https://github.com/rusefi/rusefi/files/7599272/2021-11-24_16.03.29.mlg.zip)

Seems fixed.

rusefillc, 2021-11-25:
@stefanst I bet that's not the shape you were expecting :)

rusefillc, 2021-11-25:
[image]

315-405 ratio between spikes

mck1117, 2021-11-25:
This is plotted where the X axis is the crank phase angle (0 to 360) and the Y axis is manifold pressure.  Color is to differentiate first-half vs second-half.

[image]

A good option may be to sample at a particular instant (green line), then if the pressure is high, we're on "side A", and if low we're on "side B".  Much much much simpler than trying to do peak detection.

mck1117, 2021-11-25:
green: sampling instants (configured)
red: threshold (configured)
blue: synthetic "cam signal"

[image]

rusefillc, 2021-11-30:
@NMSTEC can I please have instant MAP data log in text form?

[image]

mck1117, 2021-11-30:
[map_instant.csv](https://github.com/rusefi/rusefi/files/7628496/map_instant.csv)

rusefillc, 2021-12-03:
@NMSTEC I've just added 
TEMPLOG_MAP_INSTANT_AVERAGE 
and
TEMPLOG_map_peak
logic into firmware, can you please update in 20 minutes from now and make a new cranking log? and post it as long as TEMPLOG_MAP_INSTANT_AVERAGE  is there?

NMSTEC, 2021-12-03:
https://rusefi.com/online/view.php?log=469

rusefillc, 2021-12-04:
These fields are only logged if V-twin VVT is selected

[image]

also please select dummy "digital 4" for CAM input

rusefillc, 2021-12-04:
@NMSTEC thank you for amazing test data

At the moment we use average(length=8) to locate pressure peaks and that's not enough smoothing as seen on your data. Need a longer averaging count

https://rusefi.com/online/view.php?log=470

[image]

rusefillc, 2021-12-04:
https://rusefi.com/online/view.php?log=472

[image]

NMSTEC, 2021-12-05:
Sweet baby Cheesas! length=16, skip ratio=50, useLowRise=true (useFall?)

```
2021-12-05_13_23_10_071: EngineState: VVT1 rpm=317 time=285 eventIndex=3 gapIndex=1: gap=1.083 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_178: EngineState: VVT1 rpm=317 time=285 eventIndex=1 gapIndex=0: gap=1.000 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_178: EngineState: VVT1 rpm=317 time=285 eventIndex=1 gapIndex=1: gap=0.923 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_394: EngineState: VVT1 rpm=317 time=285 eventIndex=3 gapIndex=0: gap=0.944 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_395: EngineState: VVT1 rpm=317 time=285 eventIndex=3 gapIndex=1: gap=1.000 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_504: EngineState: VVT1 rpm=317 time=285 eventIndex=1 gapIndex=0: gap=0.970 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_504: EngineState: VVT1 rpm=317 time=285 eventIndex=1 gapIndex=1: gap=0.944 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_613: EngineState: VVT1 rpm=317 time=285 eventIndex=3 gapIndex=0: gap=0.393 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_614: EngineState: VVT1 rpm=317 time=285 eventIndex=3 gapIndex=1: gap=0.970 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_614: EngineState: VVT1 rpm=317 time=285 eventIndex=5 gapIndex=0: gap=0.307 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_614: EngineState: VVT1 rpm=317 time=285 eventIndex=5 gapIndex=1: gap=0.393 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_614: EngineState: VVT1 rpm=317 time=285 eventIndex=7 gapIndex=0: gap=0.499 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_614: EngineState: VVT1 rpm=317 time=285 eventIndex=7 gapIndex=1: gap=0.307 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_615: EngineState: VVT1 rpm=317 time=285 eventIndex=9 gapIndex=0: gap=1.000 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_617: EngineState: VVT1 rpm=317 time=285 eventIndex=9 gapIndex=1: gap=0.499 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_720: EngineState: VVT1 rpm=317 time=285 eventIndex=11 gapIndex=0: gap=3.499 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_721: EngineState: VVT1 rpm=317 time=285 eventIndex=11 gapIndex=1: gap=1.000 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_721: EngineState: VVT1 rpm=317 time=285 eventIndex=13 gapIndex=0: gap=1.000 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_721: EngineState: VVT1 rpm=317 time=285 eventIndex=13 gapIndex=1: gap=3.499 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_721: EngineState: VVT1 rpm=317 time=285 eventIndex=15 gapIndex=0: gap=0.999 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_722: EngineState: VVT1 rpm=317 time=285 eventIndex=15 gapIndex=1: gap=1.000 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_722: EngineState: VVT1 rpm=317 time=285 eventIndex=17 gapIndex=0: gap=0.571 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_722: EngineState: VVT1 rpm=317 time=285 eventIndex=17 gapIndex=1: gap=0.999 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_830: EngineState: VVT1 rpm=317 time=285 eventIndex=19 gapIndex=0: gap=1.750 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_830: EngineState: VVT1 rpm=317 time=285 eventIndex=19 gapIndex=1: gap=0.571 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_830: EngineState: VVT1 rpm=317 time=285 eventIndex=21 gapIndex=0: gap=0.571 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_830: EngineState: VVT1 rpm=317 time=285 eventIndex=21 gapIndex=1: gap=1.750 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_831: EngineState: VVT1 rpm=317 time=285 eventIndex=23 gapIndex=0: gap=0.750 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_831: EngineState: VVT1 rpm=317 time=285 eventIndex=23 gapIndex=1: gap=0.571 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_831: EngineState: VVT1 rpm=317 time=285 eventIndex=25 gapIndex=0: gap=0.999 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_831: EngineState: VVT1 rpm=317 time=285 eventIndex=25 gapIndex=1: gap=0.750 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_939: EngineState: VVT1 rpm=317 time=285 eventIndex=27 gapIndex=0: gap=1.666 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_939: EngineState: VVT1 rpm=317 time=285 eventIndex=27 gapIndex=1: gap=0.999 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_939: EngineState: VVT1 rpm=317 time=285 eventIndex=29 gapIndex=0: gap=1.799 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_940: EngineState: VVT1 rpm=317 time=285 eventIndex=29 gapIndex=1: gap=1.666 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_940: EngineState: VVT1 rpm=317 time=285 eventIndex=31 gapIndex=0: gap=0.444 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_940: EngineState: VVT1 rpm=317 time=285 eventIndex=31 gapIndex=1: gap=1.799 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_10_940: EngineState: VVT1 rpm=317 time=285 eventIndex=33 gapIndex=0: gap=1.250 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_940: EngineState: VVT1 rpm=317 time=285 eventIndex=33 gapIndex=1: gap=0.444 expected from 0.967 to 1.612 error=No
2021-12-05_13_23_11_047: EngineState: VVT1 rpm=317 time=285 eventIndex=35 gapIndex=0: gap=0.800 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_11_047: EngineState: VVT1 rpm=317 time=285 eventIndex=35 gapIndex=1: gap=1.250 expected from 0.967 to 1.612 error=No
```

rusefillc, 2021-12-05:
We still have work to do, I do not see reliable pattern :(

```
2021-12-05_13_23_10_178: EngineState: VVT1 rpm=317 time=285 eventIndex=1 gapIndex=0: gap=1.000 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_394: EngineState: VVT1 rpm=317 time=285 eventIndex=3 gapIndex=0: gap=0.944 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_504: EngineState: VVT1 rpm=317 time=285 eventIndex=1 gapIndex=0: gap=0.970 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_613: EngineState: VVT1 rpm=317 time=285 eventIndex=3 gapIndex=0: gap=0.393 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_614: EngineState: VVT1 rpm=317 time=285 eventIndex=5 gapIndex=0: gap=0.307 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_614: EngineState: VVT1 rpm=317 time=285 eventIndex=7 gapIndex=0: gap=0.499 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_615: EngineState: VVT1 rpm=317 time=285 eventIndex=9 gapIndex=0: gap=1.000 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_720: EngineState: VVT1 rpm=317 time=285 eventIndex=11 gapIndex=0: gap=3.499 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_721: EngineState: VVT1 rpm=317 time=285 eventIndex=13 gapIndex=0: gap=1.000 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_721: EngineState: VVT1 rpm=317 time=285 eventIndex=15 gapIndex=0: gap=0.999 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_722: EngineState: VVT1 rpm=317 time=285 eventIndex=17 gapIndex=0: gap=0.571 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_830: EngineState: VVT1 rpm=317 time=285 eventIndex=19 gapIndex=0: gap=1.750 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_830: EngineState: VVT1 rpm=317 time=285 eventIndex=21 gapIndex=0: gap=0.571 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_831: EngineState: VVT1 rpm=317 time=285 eventIndex=23 gapIndex=0: gap=0.750 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_831: EngineState: VVT1 rpm=317 time=285 eventIndex=25 gapIndex=0: gap=0.999 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_939: EngineState: VVT1 rpm=317 time=285 eventIndex=27 gapIndex=0: gap=1.666 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_939: EngineState: VVT1 rpm=317 time=285 eventIndex=29 gapIndex=0: gap=1.799 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_940: EngineState: VVT1 rpm=317 time=285 eventIndex=31 gapIndex=0: gap=0.444 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_10_940: EngineState: VVT1 rpm=317 time=285 eventIndex=33 gapIndex=0: gap=1.250 expected from 0.584 to 0.974 error=No
2021-12-05_13_23_11_047: EngineState: VVT1 rpm=317 time=285 eventIndex=35 gapIndex=0: gap=0.800 expected from 0.584 to 0.974 error=No
```

Note lack of clear sequence of ratios like for instance 0.4/1.2/0.4/1.2

rusefillc, 2021-12-07:
Let's pause peak-based approach and try the "compare MAP at magic angle" approach

rusefillc, 2021-12-10:
@NMSTEC new gauge "TEMPLOG_MAP_AT_CYCLE_COUNTGauge"

your goal is for that TEMPLOG_MAP_AT_CYCLE_COUNTGauge to always show "2" please find magic angle

NMSTEC, 2021-12-10:
[DataLogs.zip](https://github.com/rusefi/rusefi/files/7695522/DataLogs.zip)
0-360 in 10* increments. Using gauge shows all suck. Checked a few logs, and gauge was lying. according to gauge 230* was best

mck1117, 2021-12-11:
[image]

NMSTEC, 2021-12-11:
Note: Needs to be disabled after startup

rusefillc, 2021-12-11:
Status update: @NMSTEC reports success with "another" mode, we shall remove the first attempt with peaks at some point.

rusefillc, 2021-12-11:
@NMSTEC just limited to below cranking RPM

commit 4292c2fb (2021-12-17): MAP phase sensing fix #3544

removing dead implementation
closed 2021-12-17 by rusefillc