rusefi/rusefi #3558: Startup Pin State Error

stefanst, 2021-11-16:
[image]

This is on a frankenso board on an NB1 Miata. The pin is PE8 for tach-out. This did not happen with FW from August 2020. Current FW is from 2021/10/09
This error happens immediately on startup. Assigning a different pin for the same function removes the error. Setting the pin to "open collector" instead of "default" removes the error. @mck1117 claims this is because the test is not performed for "open collector". 
Further testing show that the error also occurs with the discovery board removed from the frankenso, indicating that it's more firmware than hardware related.

rusefillc, 2021-11-16:
@stefanst next step is checking discovery schematics and look if discovery has any funny busines on PE8

remember that discovery has a ton of random crap on it

rusefillc, 2021-11-16:
I am unable to find anything on PE8

@stefanst is this locally at your place? can you check PE8 voltage with DMM while not on Frankenso?

stefanst, 2021-11-16:
I can't find any funny business on PE8 either.
And yes- I have the unit at home. I will check tonight. What is expected?

rusefillc, 2021-11-16:
@stefanst looks like we need to improve the error message

this is about rusEFI code reading actual pin and finding high voltage when low voltage was expected. i.e. hardware self-validation in terms of pull-up/pull-down etc

Q1: did you know that or is the message super unclear?
Q2: how shall we improve the message?

as for original question - DMM to check difference between PE8 and "good" pin on start-up

stefanst, 2021-11-17:
It was a defective Discovery board. Pin PE8 shows 2.8V on startup. A different Discovery works without issues.

On the warning:
Error messages always assume some prior knowledge on the part of the consumer of said error message, or at least a way to look things up. Error messages can't go into all the detail necessary for any user to understand what's going on. 
That being said: "Actual value = 1, Logical Value = 1" does not sound like an error to me, even knowing now that this an error generated by a startup check. 1 and 1 seem fine.
Also: the initial "You have a settings error!" lead me to believe that this is firmware/settings related and not hardware related.

mck1117, 2021-11-17:
This is now TWO failures we've seen of the STM32 pin connected to a 12v-powered TC4427. @BeerMoneyMotorsports is shipping me a failed miata hellen with the same issue, also on a pin connected to a 12v output driver.

stefanst, 2021-11-17:
This particular Discovery and the associated frankenso have been knocking about, unprotected, in various plastic bins and have moved house with me twice (I think). Then they have been shipped by non-professionals, used in modified cars etc. I would not automatically assume that the problem with PE8 is caused by design flaws. Operator error is a strong possibility.

rusefillc, 2021-11-17:
@jharvey @blundar @andreika-git any ideas on this as "hardware damage to stm32 if +12v TC4427"?

Qwerty-OFF, 2021-11-17:
Hellen One boards do not have these resistors:
[image]
But if the problem exists in Frankenso, then the problem is not due to these resistors.
Is a tachometer an inductive load? Maybe we need a flyback diode.

stefanst, 2021-11-18:
[image]
[image]
Not sure what the board revision is, but here are images that may help in figuring it out.
Please note that the blown Discovery was very likely in several different Frankenso boards, since I used to swap freely between them. The problem may have existed for a long time and gone undetected. Therefore, the information which board it resided in last may not be very useful.

rusefillc, 2021-11-18:
this is latest version of power supply, that looks like latest HW revision to me.

jharvey, 2021-11-22:
The REV is behind the inductor in the picture. 
[image]
I see hand markings of "82" would that serial number tell us what rev it is?

rusefillc, 2021-11-22:
Realistically we only care if this is before 0.4 or not. We see round inductors so this board is at least 0.5 meaning latest TC4427 configuration

rusefillc, 2021-11-27:
@mck1117 @stefanst what shall we do about this ticket? close in light of https://github.com/andreika-git/hellen-one/issues/118 ?

stefanst, 2021-11-27:
Sorry- this is beyond my pay grade. Over to @mck1117

stefanst, 2021-11-29:
Well, the owner received the frankenso back. Plugged it in, and the PE8 error is back. So we fried another discovery board.

rusefillc, 2021-11-29:
Great news we can reproduce this cheaply!!!

@Qwerty-OFF @jharvey @BeerMoneyMotorsports @mck1117 @andreika-git need your help here I am totally out of my domain.

is tachometer inductive load or not? what kind of load is tachometer on this vehicle?

stefanst, 2021-11-29:
The tach output feeds into a small computer in the cluster on NB Miatas IIRC. I doubt it's inductive. 
I have a spare cluster and can do some digging tonight.

mck1117, 2021-11-29:
does this guy have a multimeter handy?

rusefillc, 2022-05-01:
@stefanst @mck1117 what is the state here? did the guy give up or else?