rusefi/rusefi #3559: Soft Brick H64b

BeerMoneyMotorsports, 2021-11-16:
Hellen H64b seems to have bricked while user was installing.  Further investigation required.  I am sending a replacement and getting this unit back for root cause analysis.

https://rusefi.com/forum/viewtopic.php?f=19&t=2212&p=43224#p43224

BeerMoneyMotorsports, 2021-12-02:
Board received. 

1- all 4 lights come on dimly but go out. 
2- no comms with pc. Including while attempting manual dfu
3- stm32f4 module shows signs of damage
[image]
[image]

rusefillc, 2021-12-02:
@Qwerty-OFF any ideas?

BeerMoneyMotorsports, 2021-12-02:
Update-  checked and found that component has lifted from the board. I removed it, cleaned and reinstalled.  Board now communicates as expected. 

I need to know if connecting pin 1N(tps 5v) direct to ground would lead to this failure.  i believe customer installed ecu with stock non variable tps still connected.

BeerMoneyMotorsports, 2021-12-02:
[image]