rusefi/rusefi #3561: CI lua in simulator

rusefillc, 2021-11-16:
Do we have Lua in simulator? Do we have CI asserting that Lua still runs under simulator?

[image]

[image]

rusefillc, 2021-11-16:
@mck1117 can you help? can you reproduce the issue? it looks like simulator is supposed to have Lua?

mck1117, 2021-11-16:
Simulator shouuuuuld have Lua.  If it doesn't, that's a bug :)

rusefillc, 2021-11-16:
'hellolua' .. '0' asserting we get hellolua0 ?

mck1117, 2021-11-16:
simulator stdout shows that it loaded lua:

[image]

rusefillc, 2021-11-16:
does rusEFI console respond to Lua commands?

mck1117, 2021-11-17:
oh, I was testing old bits, it in fact DOES NOT work in master simulator

rusefillc, 2021-11-17:
We've fixed the bug, reopening for CI

PR #3563 (2021-11-17): simulator gets a heap
If you give the Lua interpreter no memory, it does not work.

fix #3561