rusefi/rusefi #3566: Something is still odd with some F7 boards  - bricking on incremental flash

rusefillc, 2021-11-17:
on some/many F7 units incremental DFU flash via command line stm32 causes bricking

only full erase and flashing of same helps

rusefillc, 2021-11-17:
https://www.rusefi.com/forum/viewtopic.php?p=43239#p43239 has a failing dump

one day someone should try bricking by flashing same exact firmware again, maybe after changing settings?

i.e.
step 1) erase and flash of release
step 2) change settings
step 3) incremental flash of same exact release

rusefillc, 2021-11-21:
https://github.com/rusefi/rusefi/wiki/F7-requires-full-erase

NMSTEC, 2021-11-23:
Good news, I might as well as I have an STLink already connected.

NMSTEC, 2021-11-27:
Having issues replicating this with ST-Link only. Should I try bricking it via console? Boots everytime normally. No file corruption.
[image]

NMSTEC, 2021-11-27:
[image]

NMSTEC, 2021-11-27:
Attempted to change plenty of settings that require reboot on ECU. Then reconnect, confirm ok. shutdown TS, AutoUpdate console, works fine.
No flash differences detected. Size 0x8399C

Recommend checking if OB options are set to dual bank.

[image]

NMSTEC, 2021-11-27:
Tried single banking it. Still works fine. TS also detects that, thats good.

mck1117, 2021-11-29:
@NMSTEC saw

[image]

mck1117, 2021-11-29:
0x0808 0008 is just after 0x0808 0000, which is the start of sector 12, the start of the second flash bank.

[image]

I wonder if this is a bug in STM32CubeProgrammer where it has trouble doing a contiguous erase+write across the bank boundary.

mck1117, 2021-11-29:
https://community.st.com/s/question/0D53W00001EEBXXSA5/stm32cubeprogrammer-incorrectly-erases-on-dual-bank-1mb-stm32f7

rusefillc, 2021-12-06:
Lets report at https://github.com/STMicroelectronics/STM32CubeF7

rusefillc, 2021-12-16:
Also https://forum.chibios.org/viewtopic.php?f=16&t=5988

NMSTEC, 2022-01-07:
PLEASE IGNORE THIS. THIS IS NOTES FOR MYSELF TILL I RECHECK TONIGHT.
[image]
Does not work on this CPU.

rusefillc, 2023-01-11:
let's see if latest STM32Cube v2.12 is still affected?
https://github.com/rusefi/rusefi/commit/e67c2e85e1d2ea34a9527f3243cfb2d79ceee0dd

rusefillc, 2023-01-11:
```
Console version 20221225
Windows 10.0
Bundle rusefi_bundle_proteus_f7
Auto-detecting port...

Reboot command [reboot_dfu] sent into serial=COM5!

Detected rusEFI on COM5

Giving time for USB enumeration...
Executing wmic path win32_pnpentity where "Caption like '%STM32%' and Caption like '%Bootloader%'" get Caption,ConfigManagerErrorCode /format:list

Caption=STM32  BOOTLOADER

ConfigManagerErrorCode=0

Done!
Caption=STM32  BOOTLOADERConfigManagerErrorCode=0

Executing .\STM32_Programmer_CLI/bin/STM32_Programmer_CLI.exe -c port=usb1 -w "C:\stuff\bundles\rusefi_bundle_proteus_f7\snapshot_20221208_183736_proteus_f7_rusefi\console\..\rusefi.bin" 0x08000000 -v -s
      -------------------------------------------------------------------
                       STM32CubeProgrammer v2.12.0                  
      -------------------------------------------------------------------

USB speed   : Full Speed (12MBit/s)
Manuf. ID   : STMicroelectronics
Product ID  : STM32  BOOTLOADER
SN          : 2067385D5948
DFU protocol: 1.1
Board       : --
Device ID   : 0x0451
Device name : STM32F76x/STM32F77x
Flash size  : 2 MBytes (default)
Device type : MCU
Revision ID : --  
Device CPU  : Cortex-M7

Memory Programming ...
Opening and parsing file: rusefi.bin
  File          : rusefi.bin
  Size          : 513.66 KB 
  Address       : 0x08000000 

Erasing memory corresponding to segment 0:
Erasing internal memory sectors [0 8]
erasing sector 0000 @: 0x08000000 done
erasing sector 0001 @: 0x08004000 done
erasing sector 0002 @: 0x08008000 done
erasing sector 0003 @: 0x0800c000 done
erasing sector 0004 @: 0x08010000 done
erasing sector 0005 @: 0x08020000 done
erasing sector 0006 @: 0x08040000 done
erasing sector 0007 @: 0x08060000 done
erasing sector 0008 @: 0x08080000 done
Download in Progress:
±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±± 0%
  2%  4%  6%  8% 10% 12% 14% 16% 18% 20% 22% 24% 26% 28% 30% 32% 34% 36% 38% 40% 42% 44% 46% 48% 50% 52% 54% 56% 58% 60% 62% 64% 66% 68% 70% 72% 74% 76% 78% 80% 82% 84% 86% 88% 90% 92% 94% 96% 98% 100%

File download complete
Time elapsed during download operation: 00:00:19.281

Verifying ...

Read progress:
±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±± 50%
  2%  4%  6%  8% 10% 12% 14% 16% 18% 20% 22% 24% 26% 28% 30% 32% 34% 36% 38% 40% 42% 44% 46% 48% 50% 52% 54% 56% 58% 60% 62% 64% 66% 68% 70% 72% 74% 76% 78% 80% 82% 84% 86% 88% 90% 92% 94% 96% 98% 100%

Download verified successfully 

RUNNING Program ... 
  Address:      : 0x8000000
Start operation achieved successfully
Done!
SUCCESS!
Please power cycle device to exit DFU mode

```

rusefillc, 2023-01-11:
one F7 board seems to have worked for me now with above-512Kb firmware

@mck1117 @NMSTEC what would be a proper test to test/close this issue?

NMSTEC, 2023-01-11:
I have an F7 Proteus that needs fresh fw. Can attempt automatic update.

rusefillc, 2023-01-22:
@mck1117 @NMSTEC

mck1117, 2023-01-22:
Looks like this is NOT fixed

[image]

rusefillc, 2023-01-22:
@mck1117 can you try rusefi console command? Is it not verifying the same?

mck1117, 2023-01-22:
broken with console too

looks like we have to go back to 512k image on f7

[image]

rusefillc, 2023-01-22:
We would have to fit into 512kb somehow

rusefillc, 2023-01-22:
Fun fact DFU protocol pretty trivial worst case we can just java dfu

mck1117, 2023-01-22:
> We would have to fit into 512kb somehow

currently only ~3k over

rusefillc, 2023-01-22:
Even when we shrink it's nice to be able to add stuff right?

I am shocked that it's not resolved by ST still

rusefillc, 2023-01-30:
failed for me as well

```
Console version 20230107
Windows 10.0
Bundle rusefi.snapshot.proteus_f7
Auto-detecting port...

Reboot command [reboot_dfu] sent into serial=COM45!

Detected rusEFI on COM45

Giving time for USB enumeration...
Executing wmic path win32_pnpentity where "Caption like '%STM32%' and Caption like '%Bootloader%'" get Caption,ConfigManagerErrorCode /format:list

Caption=STM32  BOOTLOADER

ConfigManagerErrorCode=0

Done!
Caption=STM32  BOOTLOADERConfigManagerErrorCode=0

Executing .\STM32_Programmer_CLI/bin/STM32_Programmer_CLI.exe -c port=usb1 -w "C:\stuff\bundles\rusefi_bundle_proteus_f7\rusefi.snapshot.proteus_f7\console\..\rusefi.bin" 0x08000000 -v -s
      -------------------------------------------------------------------
                       STM32CubeProgrammer v2.12.0                  
      -------------------------------------------------------------------

USB speed   : Full Speed (12MBit/s)
Manuf. ID   : STMicroelectronics
Product ID  : STM32  BOOTLOADER
SN          : 346335593031
DFU protocol: 1.1
Board       : --
Device ID   : 0x0451
Device name : STM32F76x/STM32F77x
Flash size  : 2 MBytes (default)
Device type : MCU
Revision ID : --  
Device CPU  : Cortex-M7

Memory Programming ...
Opening and parsing file: rusefi.bin
  File          : rusefi.bin
  Size          : 515.80 KB 
  Address       : 0x08000000 

Erasing memory corresponding to segment 0:
Erasing internal memory sectors [0 8]
erasing sector 0000 @: 0x08000000 done
erasing sector 0001 @: 0x08004000 done
erasing sector 0002 @: 0x08008000 done
erasing sector 0003 @: 0x0800c000 done
erasing sector 0004 @: 0x08010000 done
erasing sector 0005 @: 0x08020000 done
erasing sector 0006 @: 0x08040000 done
erasing sector 0007 @: 0x08060000 done
erasing sector 0008 @: 0x08080000 done
Download in Progress:
±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±± 0%
  2%  4%  6%  8% 10% 12% 14% 16% 18% 20% 22% 24% 26% 28% 30% 32% 34% 36% 38% 40% 42% 44% 46% 48% 50% 52% 54% 56% 58% 60% 62% 64% 66% 68% 70% 72% 74% 76% 78% 80% 82% 84% 86% 88% 90% 92% 94% 96% 98% 100%

File download complete
Time elapsed during download operation: 00:00:18.645

Verifying ...

Read progress:
±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±±± 50%
  2%  4%  6%  8% 10% 12% 14% 16% 18% 20% 22% 24% 26% 28% 30% 32% 34% 36% 38% 40% 42% 44% 46% 48% 50% 52% 54% 56% 58% 60% 62% 64% 66% 68% 70% 72% 74% 76% 78% 80% 82% 84% 86% 88% 90% 92% 94% 96% 98% 100%

Error: Data mismatch found at address  0x08080000 (byte = 0x41 instead of 0xD1)

Error: Download verification failed

Done!
1923 / 0
ERROR: does not look like DFU has worked!
Devices: OK         USBDevice       STM32  BOOTLOADER                                                                USB\VID_...
hint: error state is already in your clipboard, please use PASTE or Ctrl-V while reporting issues

```

rusefillc, 2023-01-30:
Poking on https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer/issues/19

rusefillc, 2023-01-30:
Nagging https://community.st.com/s/question/0D53W0000269AEMSA2/stm32-cube-bug-working-with-stm32f767zg-first-reported-in-2021-how-to-get-progress

rusefillc, 2023-01-30:
More nagging https://forum.chibios.org/viewtopic.php?f=16&t=6186

rusefillc, 2023-01-30:
New Internal ticket number: 144023

rusefillc, 2023-02-27:
#5135

rusefillc, 2024-01-30:
Got some warm words at https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer/issues/19#issuecomment-1916625922

rusefillc, 2024-03-21:
they have released new #6274

rusefillc, 2024-07-09:
took ST less than 4 years to fix so who are we complain?

rusefillc, 2024-07-09:
or not?

[image]

rusefillc, 2024-07-09:
> If you flash with openblt as the normal thing to do, this isn't an issue.

openblt is definitely a great work-around, the priority of the issue here is low. Mostly my OCD demanding that I know if ST has fixed or not.

rusefillc, 2024-07-30:
https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer/issues/34 and https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer/issues/35 reported.

FDSoftware, 2024-12-18:
Reading a bit of the F7 code I found this issue, I update with the alternative I found, according to dfu-util, ST's DFU is incompatible with DFU 1.1, they use their own protocol called DfuSe
At the same time, dfu-util supports flashing on F2/F3/F4 targets by adding a flag:
https://dfu-util.sourceforge.net/dfuse.html

But according to: "USB DFU protocol used in the STM32 bootloader" AN3156
https://www.st.com/resource/en/application_note/cd00264379-usb-dfu-protocol-used-in-the-stm32-bootloader-stmicroelectronics.pdf

The same protocol is also used in F7/H7, so as an alternative to STM32CubeProgrammer we could try dfu-util for all the targets we currently support