rusefi/rusefi #3581: On -fpermissive in Windows in unit tests

rusefillc, 2021-11-20:
```
C:\stuff\rusefi\firmware>gcc -v                                                                                                                              Using built-in specs.                                                                                                                                        COLLECT_GCC=/usr/bin/gcc                                                                                                                                     COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-cygwin/11/lto-wrapper.exe                                                                                         Target: x86_64-pc-cygwin                                                                                                                                     Configured with: /mnt/share/cygpkgs/gcc/gcc.x86_64/src/gcc-11.2.0/configure --srcdir=/mnt/share/cygpkgs/gcc/gcc.x86_64/src/gcc-11.2.0 --prefix=/usr --exec-prefix=/usr --localstatedir=/var --sysconfdir=/etc --docdir=/usr/share/doc/gcc --htmldir=/usr/share/doc/gcc/html -C --build=x86_64-pc-cygwin --host=x86_64-pc-cygwin --target=x86_64-pc-cygwin --without-libiconv-prefix --without-libintl-prefix --libexecdir=/usr/lib --with-gcc-major-version-only --enable-shared --enable-shared-libgcc --enable-static --enable-version-specific-runtime-libs --enable-bootstrap --enable-__cxa_atexit --with-dwarf2 --with-tune=generic --disable-bootstrap --enable-languages=c,c++,fortran,lto,objc,obj-c++,jit --enable-graphite --enable-threads=posix --enable-libatomic --enable-libgomp --enable-libquadmath --enable-libquadmath-support --disable-libssp --enable-libada --disable-symvers --with-gnu-ld --with-gnu-as --with-cloog-include=/usr/include/cloog-isl --without-libiconv-prefix --without-libintl-prefix --with-system-zlib --enable-linker-build-id --with-default-libstdcxx-abi=gcc4-compatible --enable-libstdcxx-filesystem-ts                                                                                                                                              Thread model: posix                                                                                                                                          Supported LTO compression algorithms: zlib zstd                                                                                                              gcc version 11.2.0 (GCC)                                                                                                                                                                                                                                                                                                  C:\stuff\rusefi\firmware>i686-w64-mingw32-gcc -v                                                                                                             Using built-in specs.                                                                                                                                        COLLECT_GCC=/usr/bin/i686-w64-mingw32-gcc                                                                                                                    COLLECT_LTO_WRAPPER=/usr/lib/gcc/i686-w64-mingw32/11/lto-wrapper.exe                                                                                         Target: i686-w64-mingw32                                                                                                                                     Configured with: /mnt/share/cygpkgs/mingw64-i686-gcc/mingw64-i686-gcc.x86_64/src/gcc-11.2.0/configure --srcdir=/mnt/share/cygpkgs/mingw64-i686-gcc/mingw64-i686-gcc.x86_64/src/gcc-11.2.0 --prefix=/usr --exec-prefix=/usr --localstatedir=/var --sysconfdir=/etc --docdir=/usr/share/doc/mingw64-i686-gcc --htmldir=/usr/share/doc/mingw64-i686-gcc/html -C --build=x86_64-pc-cygwin --host=x86_64-pc-cygwin --target=i686-w64-mingw32 --without-libiconv-prefix --without-libintl-prefix --with-sysroot=/usr/i686-w64-mingw32/sys-root --with-build-sysroot=/usr/i686-w64-mingw32/sys-root --disable-multilib --disable-win32-registry --enable-languages=c,c++,fortran,lto,objc,obj-c++ --enable-fully-dynamic-string --enable-graphite --enable-libgomp --enable-libquadmath --enable-libquadmath-support --enable-libssp --enable-version-specific-runtime-libs --with-dwarf2 --with-gnu-ld --with-gnu-as --with-gcc-major-version-only --with-tune=generic --with-cloog-include=/usr/include/cloog-isl --with-system-zlib --enable-threads=posix --libexecdir=/usr/lib                                                               Thread model: posix                                                                                                                                          Supported LTO compression algorithms: zlib zstd                                                                                                              gcc version 11.2.0 (GCC)                                                                                                                                                                                                                                                                                                  C:\stuff\rusefi\firmware>                       
```

rusefillc, 2021-11-20:
[image]

rusefillc, 2021-11-20:
is this a 32 bit / 64 bit question or not at all?

```
# It looks like cygwin build of mingwg-w64 has issues with gcov runtime :(
# mingw-w64 is a project which forked from mingw in 2007 - be careful not to confuse these two.
# In order to have coverage generated please download from https://mingw-w64.org/doku.php/download/mingw-builds
# Install using mingw-w64-install.exe instead of similar thing packaged with cygwin
# Both 32 bit and 64 bit versions of mingw-w64 are generating coverage data.

ifeq ($(OS),Windows_NT)
ifeq ($(USE_MINGW32_I686),)
#this one is 64 bit
  TRGT = x86_64-w64-mingw32-
else
#this one was 32 bit
  TRGT = i686-w64-mingw32-
endif
else
  TRGT = 
endif
```

rusefillc, 2021-11-20:
and failure is
```
Compiling signal_executor_sleep.cpp
Compiling single_timer_executor.cpp
Compiling pwm_generator_logic.cpp
Compiling event_queue.cpp
../firmware/controllers/system/timer/pwm_generator_logic.cpp: In member function 'void PwmConfig::handleCycleStart()':
../firmware/controllers/system/timer/pwm_generator_logic.cpp:151:84: warning: comparison of integer expressions of different signedness: 'int' and 'uint32_t' {aka 'unsigned int'} [-Wsign-compare]
  151 |                 if (forceCycleStart || safe.periodNt != periodNt || safe.iteration == iterationLimit) {
      |                                                                     ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
../firmware/controllers/system/timer/event_queue.cpp: In member function 'bool EventQueue::executeOne(efitime_t)':
../firmware/controllers/system/timer/event_queue.cpp:215:58: error: cast from 'scheduling_s*' to 'long int' loses precision [-fpermissive]
  215 |         printf("QUEUE: execute current=%d param=%d\r\n", (long)current, (long)action.getArgument());
      |                                                          ^~~~~~~~~~~~~
../firmware/controllers/system/timer/event_queue.cpp:215:73: error: cast from 'void*' to 'long int' loses precision [-fpermissive]
  215 |         printf("QUEUE: execute current=%d param=%d\r\n", (long)current, (long)action.getArgument());
      |                                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~
Compiling settings.cpp
Compiling error_handling.cpp
Compiling map_averaging.cpp
Compiling high_pressure_fuel_pump.cpp
Compiling rpm_calculator.cpp
```

mck1117, 2021-11-20:
oh, that might be a 32bit thing.  Can you try casting to `uintptr_t` instead of long?

racer-coder, 2021-11-20:
bitten by Windows not long long

commit fcc2b780 (2021-11-20): On -fpermissive in Windows in unit tests fix #3581
closed 2021-11-20 by rusefillc