rusefi/rusefi #3591: SD card logs

rusefillc, 2021-11-22:
Martin reports that SD card logs do not open with MLV

rusefillc, 2021-11-22:
[re_14.zip](https://github.com/rusefi/rusefi/files/7583713/re_14.zip)

mck1117, 2021-11-22:
that file contains a valid header and no data

mck1117, 2021-11-22:
actually, it contains a nearly valid header (missing the last 182 bytes it seems)

rusefillc, 2021-11-22:
Do we have a defect of some sort that we do not log anything?

I assume we are supposed to log without RPM

rusefillc, 2021-11-22:
Do we overflow silently now with all the fancy extra fields?

mck1117, 2021-11-22:
works on f4

rusefillc, 2021-11-22:
ohh even worse DMA drama?!

mck1117, 2021-11-22:
H7 test 1: writing some files via MSD then reading them back after MCU reset results in same checksum -> H7 MSD works fine
H7 test 2: log file written to SD card is valid

mck1117, 2021-11-22:
**F7** test 1: writing some files via MSD then reading them back after MCU reset results in same checksum -> F7 MSD works fine

mck1117, 2021-11-22:
> Do we overflow silently now with all the fancy extra fields?

yes 😢

mck1117, 2021-11-22:
OK, so this is a buffer overflow.  On F4 we're lucky that it happens to overflow in to a buffer that isn't used when the PC isn't connected, aka all the time that we'd be logging, so it's PROBABLY safe.  H7 and F7 it gets stomped by another buffer which breaks SD logging.

mck1117, 2021-11-22:
fixed by #3596

closed 2021-11-22 by mck1117