rusefi/rusefi #3592: LUA CANbus expect 29bit ID by default

mmatousek89, 2021-11-22:
voltageFromCan = 10.5
canRxAdd(0x00000600)
setTickRate(1)

function onCanRx(bus, id, dlc, data)
  print('got CAN id=' .. id .. ' dlc='  .. dlc)
  voltageFromCan =   data[2] *256.0 + data[1]
end

function onTick()
 print('potVal: ' .. voltageFromCan)
end

this is example which works as expected 

canRxAdd(0x600)

this is from example on forum and it can evoke 11bit can id which dont work

mmatousek89, 2021-11-23:
i was expecting 11bit standard id so i setup my can transmiter to send 11bit id message

lua expect 29bit extended id so it recieve nothing

when i change transmiter to 29bit id everything works

we need to declare somewhere that lua expect 29bit id, and give user posibility to switch to 11bit id somehow, o will prefer 11bit to be default

rusefillc, 2021-11-23:
@mck1117 is this about SID vs EID? 

shall we do
```
if (CAN_SID(frame) == luaCanRxIds[i] || CAN_EID(frame) == luaCanRxIds[i]) {
```

?

mck1117, 2021-11-28:
I think this is probably fixed now...

- user can pick 11 or 29b for broadcast
- lua tx can pick per-sent-frame
- lua rx listens to both (if you're sending the same ID in both 11 and 29 on the same bus, that's wrong, so we don't really care about differentiating)

closed 2021-11-28 by mck1117