rusefi/rusefi #3593: SDcard log over ignition ON / OFF / ON with USB plugged

mmatousek89, 2021-11-22:
when im in car with laptop, and turn key off, there is no logging to sd card when i turn key back on, because ECU connected via usb go to MSD mode and key on does not cancel MSD mode

rusefillc, 2021-11-22:
@mmatousek89 please use "current behavour" and "expected behaviour" to make it a bit less confusing

In your description there

mmatousek89, 2021-11-22:
mine does not need reboot, as soon as i connect computer it goes to MSD mode

mck1117, 2021-11-22:
correct:
- boot with USB connected -> immediate MSD
- boot without USB connected -> no MSD, write log to SD

mmatousek89, 2021-11-22:
haa 2disk drives are showing up....

my bad that i named my sd card RUSEFI when formated  :-D

because now booth drives are RUSEFI ... 

one is SDcard

and one is  rusefi internal....

mmatousek89, 2021-11-22:
ok sorry on this one,  its totaly my bad  i need to be more concentrated

closed 2021-11-22 by mmatousek89