rusefi/rusefi #3597: DFU dram on one Proteus has killed DFU on second Proteus

rusefillc, 2021-11-22:
We took Proteus 0.6 and rebooted into DFU programmatically
We have installed STM32 BOOTLOADER driver
We saw message about readout protection

[image]

we invoked relevant command -rdu

this has killed everything - this proteus no longer responds to HW reset button nor to HW DFU button, and windows shows:

[image]

we tool another unit. Proteus 0.6 runs OK but if we enter HW DFU we still see broken USB message?!

huh.

NMSTEC, 2021-11-23:
Resolved. tldr, incorrectly configured MCU. 

1. After bricking, connected to st-link. Found memory wiped. 
2. Rewrote flash. Normal mode works, DFU did not. Reset would operate properly.
3. Checked with Matt, and compared to his ECU. Found incorrect Boot addresses, as well as nDBANK option bit
4. Set nDBANK to 0, changed BOOT_ADD0 from 0xffff to 0x80 & BOOT_ADD1 from 0xffff to 0x40.
5. Mass Erase and rewrite. Problem fully resolved, DFU and normal mode operate as expected.

NMSTEC, 2021-11-23:
[image]

closed 2021-11-23 by rusefillc