rusefi/rusefi #3605: Inputs from CAN

NMSTEC, 2021-11-24:
Would like to be able to set variables via CAN.

On Harley's clutch switch info is sent from the forks via CAN, and would like it to be set in ECU from that.

rusefillc, 2021-11-24:
@mck1117 shall we make clutch a Sensor?

NMSTEC, 2021-11-24:
I'd rather have option for CAN inputs. For example on a different model of harley starter IS controlled by ECU, and starter switch is CAN based.

rusefillc, 2021-11-24:
@NMSTEC do you want to create extra work and get results later, or do you prefer to stay focused and get results sooner?

NMSTEC, 2021-11-24:
Ya gotta stop thinking everything im suggesting is 100% needed right now. 
This is for future, but will be needed. To run the bike, dont even need clutch switch as we speak anyway.

rusefillc, 2021-12-16:
@NMSTEC 
https://github.com/rusefi/rusefi/wiki/Lua-Scripting#getoutputname
https://github.com/rusefi/rusefi/wiki/Lua-Scripting#setclutchupstatevalue
https://github.com/rusefi/rusefi/wiki/Lua-Scripting#getcalibrationname

commit 99d19f2d (2021-12-16): clutch input via CAN fix #3605
closed 2021-12-16 by rusefillc