rusefi/rusefi #3607: Revisit Sensor Sniffer

rusefillc, 2021-11-25:
With 1KHz data rate we do not need Sensor Sniffer in it's current form

Sensor Sniffer would be useful only if we go with higher data rate and larger buffer maybe probably once we have DRAM?

rusefillc, 2021-11-25:
[image]

rusefillc, 2021-11-25:
Short-term we can reduce chart size but keep the code?

mck1117, 2021-11-25:
With engine phase logged, we can probably nix a few of the modes, like MAP sensing. I see no need for that if we can produce an awesome log like the one above.

mck1117, 2021-11-25:
This is pretty awesome:

[image]

rusefillc, 2021-11-25:
Sensor Sniffer would be useful for full speed ADC but that would mean a larger buffer?

mck1117, 2021-11-25:
We only need a buffer at all if we're interested in >1khz sampling. If not, TunerStudio is the buffer for us, and that has functionally infinite storage.

rusefillc, 2021-11-25:
yes, exactly. that's exactly what I am saying.

as of today we are not ready for 10KHz buffer but we probably want to keep the code.