rusefi/rusefi #3611: skipspark script kills ECU

rusefillc, 2021-11-25:
martin says

[image]

rusefillc, 2021-11-26:
I am able to reproduce the issue with MRE

I get all LEDs light up and not blink until reset

rusefillc, 2021-11-26:
[image]

rusefillc, 2021-11-26:
rand() wants to do malloc and that does not go well

[image]

rusefillc, 2021-11-26:
We now use fancy https://github.com/MersenneTwister-Lab/TinyMT

closed 2021-11-26 by rusefillc

mck1117, 2021-11-26:
> malloc and that does not go well

we....actually still have a reference to `malloc` in the firmware

it's called by `strtof` within lua: https://stackoverflow.com/questions/66145835/why-do-strtod-and-strtof-of-the-newlib-c-standard-library-implementation-use

commit bdde177e (2021-11-26): skipspark script kills ECU fix #3611