rusefi/rusefi #3614: why so many channels are hidden in debug ?

mmatousek89, 2021-11-25:
can we have all of them logged all time?

mck1117, 2021-11-25:
examples?

mmatousek89, 2021-11-25:
tps diff
wastegate dc from table
etb pri/sec diff
.... and 70others

rusefillc, 2021-11-26:
@mmatousek89 historically we were concerned about RAM and traffic

At this point I believe we can afford to have way more gauges not as debug mode.

instead of asking for "70 others" it would be best to ask specifically for the ones you need this week/next week to get results quicker in a more focused way

rusefillc, 2021-11-26:
Depends on #197

rusefillc, 2021-11-27:
@racer-coder can you help? that would mean finishing #197

racer-coder, 2021-11-27:
I agree that everything should be exposed.  I think it comes down to philosophical "how should the core of the software operate", which I'm kinda trying to tackle partly with rethinking how tables work.

Using Motec M600/800/etc software as an example,
1. First you configure all the inputs.  Inputs can be direct connected sensors or received over canbus/expansion blocks.
2. Engine management goes and computes a bunch of intermediaries.  Estimated air mass, VE after compensations, Fuel required, injector duration for fuel required, etc
3. ALL of these variables - inputs + computed variables - are available to be logged, transmitted over CAN, or to be used as inputs into other tables for more computation
4. Outputs (solenoids etc) are based on the variables + some logic

Since there are so many variables, you need a tree view to fit them so you aren't scrolling forever through some giant list.  Link actually lets you type a search phrase in the dialog box that selects variables.

So yes, there should be a centralized place to describe all variables - whether inputs or not, including units, precision, scale, # bytes, etc.  Then all the logic in the software should refer to this centralized variable list.

I think we kinda already have this for sensors, but it should be extended to everything else too.

As for logging them "all the time" or not, that comes down to how much storage we have, but I assume the MCU is powerful enough and I assume we use an external microSD card so we should have plenty of bandwidth and storage space to log everything.  Link ECU lets you select which channels you log and what refresh rate you choose for each of them.

rusefillc, 2021-11-27:
@racer-coder I think we are pretty much on the same page here

specifically the whole "live data" approach with machine readable state for all controllers seems like a way to implement. In my opinion we shall extend current flat real OutputsChannels where we COPY data from model into OutputsChannels with a synthetic/virtual OutputsChannels which is composed of list of live data models under the hood

rusefillc, 2021-11-29:
introduces #3633

mck1117, 2022-04-05:
> linked list of memory regions under the hood presented to TS as one continues buffer

This doesn't need to be a linked list - we can C++ compile-time-zero-cost-list this like we do with `engineModules`

rusefillc, 2022-04-13:
Right, I guess we can but that would impost additional constraints and conventions. Conditional compilation would get trickier, we would need to make sure that no fields are added with inheritance etc etc?

closed 2022-04-18 by rusefillc

nmschulte, 2022-12-16:
should the wiki page https://github.com/rusefi/rusefi/wiki/Debug-Mode (https://www.rusefi.com/s/debugmode) and related https://github.com/rusefi/rusefi/wiki/Debug-Fields be sunset now or does this need more burn-in/effort?

mi-hol, 2023-01-26:
> should the wiki page https://github.com/rusefi/rusefi/wiki/Debug-Mode (https://www.rusefi.com/s/debugmode) and related https://github.com/rusefi/rusefi/wiki/Debug-Fields be sunset now or does this need more burn-in/effort?

@rusefillc in case the question is answered I'd be happy to update wiki

commit f55ad6ff (2022-04-18): logging of live data structs was: data points fix #3614

we now have what Martin has asked for :)