rusefi/rusefi #3618: can we please have a runner for proteus-F7 to speed up that specific bundle

rusefillc, 2021-11-26:
At the moment proteus-F7 is the most important bundle

build queue costs us a lot of human time while active troubleshooting how can we make one build much quicker?

mck1117, 2021-11-26:
#3622 speeds up our firmware builds significantly, as it doesn't need to download any dependencies or do a full clone first.

NMSTEC, 2021-11-27:
I see it as a feature, I have enough time to go out for a smoke and think about where I screwed up in the project.

chuckwagoncomputing, 2023-01-21:
Additional discovery: according to https://github.com/orgs/community/discussions/20019#discussioncomment-4722254 if you label your runners the same as the Github-hosted runners, e.g. `ubuntu-latest`, Github will run jobs on them until they're all full, then run the rest on the Github-hosted runners. This solves the main problem with #3622.

chuckwagoncomputing, 2023-01-21:
I can confirm that this Just Works

mck1117, 2023-01-21:
> Github will run jobs on them until they're all full

that's a good finding!

chuckwagoncomputing, 2023-01-28:
Disregarding self-hosting, and back to the issue at hand:
Moving it to its own workflow would not help matters, as there is no way, as far as I can tell, to give a particular workflow priority in the queue. I will close this issue, and if we want to discuss self-hosting, that can be done elsewhere.

closed 2023-01-28 by chuckwagoncomputing