rusefi/rusefi #3637: MAP lost in Datalogging after latest snapshot 20121129@28919

NMSTEC, 2021-11-29:
Working on previous firmware from 3 days ago all logging was fine. All channels are logged. Gauges show MAP properly, and bike runs fine.

Logs show 0 MAP at all times
[DataLogs.zip](https://github.com/rusefi/rusefi/files/7621439/DataLogs.zip)
[image]
[image]

rusefillc, 2021-11-29:
@NMSTEC great find, looks like mix-up between MAP and instant MAP

mck1117, 2021-11-30:
This is still not right

[image]

NMSTEC, 2021-12-02:
Could use this fixed for tomorrow with the new Cylinder pressure testing thing.

rusefillc, 2021-12-02:
please try now - there was a chance yesterday

NMSTEC, 2021-12-02:
Confirmed fixed.

commit f5bf1432 (2021-11-29): MAP lost in Datalogging after latest snapshot 20121129@2891 fix #3637
closed 2021-12-02 by NMSTEC