rusefi/rusefi #3638: Requesting LUA Torque intervention

NMSTEC, 2021-11-30:
Harleys have traction control, it requests power reduction, can be as simple as % of ETB reduction request as long as its called, and return back to stock after request has stopped.

This is also something we'd like for two other features we plan to include, 1. Wheelie control 2. Electronic shifter (100% will be installed and controlled by rusEFI via LUA & CAN)

rusefillc, 2021-11-30:
@NMSTEC edited - one issue for one ask

rusefillc, 2021-11-30:
We already have two relevant Lua functions open question which is the best overall logic

https://github.com/rusefi/rusefi/wiki/Lua-Scripting#settimingaddangle
allows to dial down timing, use with timer

https://github.com/rusefi/rusefi/wiki/Lua-Scripting#setsparkskipratio
allows to skip some sparks, same story use with timer

NMSTEC, 2021-11-30:
Would prefer to have ETB request modifiable, as that would allow for more control with smoothness. Basically driver request TPP to be overridden via lua during request.

rusefillc, 2021-12-02:
@NMSTEC please try https://github.com/rusefi/rusefi/wiki/Lua-Scripting#setetbaddextraetb

closed 2021-12-02 by rusefillc