rusefi/rusefi #3644: Does not compile under 10

rusefillc, 2021-12-02:
Thank you for taking the time to report an issue!

To make our work easier, please provide as many details as you can:
* [ ] What hardware did you use while experiencing the issue?

alxrMironov, 2021-12-02:
I've has found, that build fails at "volatile" error. 
Since C++20 some ```volatile``` keyword use cases is deprecated and without ```Wno-error=volatile``` flag - build will fail.
```
 error: compound assignment with 'volatile'-qualified left operand is deprecated [-Werror=volatile]
```
and
```
 error: '++' expression of 'volatile'-qualified type is deprecated [-Werror=volatile]
```

With ```Wno-error=volatile```, we have myriads of warnings about "volatile" variables post-incrementation and another cases, but build runs ok.

As i undestand, without explicit standard designation, the ```GCC 9```  use some kind of C++17 with C++20 additions. BUT, this C++20 additions is also used in "rusefi" and explicity choosing of ```-std=c++17``` standard would cause another billiard of errors:
```error: default member initializers for bit-fields only available with '-std=c++2a' or '-std=gnu++2a' [-Werror]```

rusefillc, 2021-12-02:
@alxrMironov would 9 accept noerror=volatile?

alxrMironov, 2021-12-02:
As i understand ```GCC 9``` has no check on this cases. So it is not needed.

alxrMironov, 2021-12-02:
> @alxrMironov would 9 accept noerror=volatile?

No, it doesnt. "No option" error.
Also, with ```-Wno-error=volatile``` at ```GCC 10``` the build output is over-polluted by warnings.

dron0gus, 2021-12-21:
`-Wno-error=deprecated`

rusefillc, 2021-12-21:
Would it work under both 9 and 10? Please PR?

dron0gus, 2021-12-21:
Need to test with gcc 9. Will do this evening.

rusefillc, 2022-05-14:
and they have Arm GNU Toolchain Version 11.2-2022.02 waiting for us at https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/downloads