rusefi/rusefi #3660: CAN wake-up

NMSTEC, 2021-12-05:
Just to clarify Harley graceful power off/power on is done via CAN not by controlling main relay

rusefillc, 2021-12-05:
Oh shit :( Worst case we would have to implement a dedicated microcontroller just for CAN and wake up.

mck1117, 2021-12-05:
Can you enumerate the full list of what happens when you go from walking up to a parked bike, thru running engine?

rusefillc, 2021-12-06:
probably relevant https://www.st.com/resource/en/application_note/dm00096220-using-stm32f4-mcu-power-modes-with-best-dynamic-efficiency-stmicroelectronics.pdf

mck1117, 2021-12-06:
That's why I wanted clarity on what we need.  If we need to wake up for ANY packet, that's pretty easy.  If we need to only wake up for a particular packet, that's harder.

rusefillc, 2021-12-06:
step one: bike is parked for a week, things are whatever
step two: rider sits on the bike with RFID in his pocket, rider does not touch any buttons
step three: ass sensing sensor sends a CAN message to start waking up modules
if ECU does not wake up and announce itself within unknown time, dash would persist an error code with error light. 3 minute of manual fuckery to wipe out code

there are only two wires with +12 ever:
pin 67 has constant +12
pin 5 has +12 when physical switch is in IGNITION position

BCM expects CAN from ECU while no power on pin 5. Conclusion: MCU in ECU is supposed to feed itself on constant power pin 67.

@NMSTEC please correct me if any of this is not right
@NMSTEC please elaborate of shutdown procedure in terms of what CAN activity happens when

NMSTEC, 2021-12-06:
We need any packet whatsoever to wake up. As the rest of the vehicle does the same. 

Scenario 1. Walking to up to bike, nothing occurs. Turn IGN to on, and BCM wakes everyone up as well as turn "ign 12v" on. Wakes all up just by simply transmitting any packet. In this case 0x502 - 0x01(or 10? I cant recall, not at work)

Scenario 2. Sit down on bike, with everything off. BCM wakes up (Motion sensor) everyone with a 0x502 message, but keeps ign voltage off. At this point all modules are awake and broadcasting, and it WILL check for ECM message. If no ECM at this point, all modules set u0100 code.

Scenario 3. Without touching rest of bike, press any button anywhere on bike (Hazards, turn signal, cluster button, whatever) That module wakes up, transmits appropriate info that its awake and that it wants to say this button is pressed. At this point BCM decides to power up ign voltage, and everyone transmits data as normal until nothing else occurs, and BCM tells everyone to go to sleep. This is usually a 5 second delay.

The shutdown procedure again, can be easily implemented in LUA with some form of access to a FW switch. Wake up is allowed on absolutely any CAN message on bus, and is expected to wake up.

This can be proven with the easy fact of when I update proteus with any data, or unplug and plug it back in, my lua code transmission wakes up everyone, and BCM supplies 12v ign.

rusefillc, 2021-12-06:
@mck1117 @NMSTEC  do I read it right that we would need a CAN shutoff to work together with CAN wake-up? i.e. we need to make sure that we wake up during overall shutdown procedure, i.e. once we decide to suspend ECU we would need to stay alive for some time while making sure that we are CAN silent?

mck1117, 2021-12-06:
> The shutdown procedure again, can be easily implemented in LUA with some form of access to a FW switch. Wake up is allowed on absolutely any CAN message on bus, and is expected to wake up.

This makes it pretty easy to implement.

mck1117, 2021-12-06:
Please confirm:
- BCM always commands turn-off
- Presence of a single CAN frame indicate wake-up

yes?

NMSTEC, 2021-12-06:
Yes to both. Hence why in the other ticket proteus transmitting that packets angers the BCM.

NMSTEC, 2021-12-06:
> @mck1117 @NMSTEC do I read it right that we would need a CAN shutoff to work together with CAN wake-up? i.e. we need to make sure that we wake up during overall shutdown procedure, i.e. once we decide to suspend ECU we would need to stay alive for some time while making sure that we are CAN silent?

Prefer to handle that in LUA, depending on whatever new un-reverse engineered scenario I discover. 
Let LUA control shutdown, and FW wake up on any packet.

mck1117, 2021-12-06:
Yeah, so all we need is:
- Lua engine stop (already have this)
- Lua ECU-sleep-mode
- Hardware (probably no software) to wake ECU from CAN traffic

NMSTEC, 2021-12-06:
It doesn't have to go into the deepest of sleeps either, 15-25ma is an acceptable power draw.

NMSTEC, 2021-12-06:
Bike overall in sleep state draws 60ma. Majority of it being the BCM and the IMU (Motion sensor). Quite normal for cars too. Mazda/fords of the early 2005-2012s ate about 130ma. That being a defect though, the BCM consumed all of that basically.

Lower current is always better, however I am talking about just in case scenarios.

rusefillc, 2021-12-06:
@mck1117 says "just use RX pin as a wakeup pin" - that would mean at least wiring wakeup pin somewhere, @NMSTEC would have to solder a hair wire @NMSTEC can you solder a hair wire on your proteus once we have more details? :)

Still open question what SW change to park stm32 how

mck1117, 2021-12-06:
No - you don't need an extra wire.  AFAIK you can configure the STM32 to wake up on an EXTI line, so you configure the CAN RX pin as EXTI so it wakes the STM32 when traffic happens.

mck1117, 2021-12-06:
@NMSTEC can you set a "target" and "absolute maximum" power draw?

NMSTEC, 2021-12-06:
> @mck1117 says "just use RX pin as a wakeup pin" - that would mean at least wiring wakeup pin somewhere, @NMSTEC would have to solder a hair wire @NMSTEC can you solder a hair wire on your proteus once we have more details? :)
> 
> Still open question what SW change to park stm32 how

Generally speaking databus input pins are exclusively all interrupt pins. All that's required is setting that pin as an interrupt in FW. 

> @NMSTEC can you set a "target" and "absolute maximum" power draw?

Target sub 5mA, Max 25mA.

mck1117, 2021-12-06:
[image]

Standby mode draws very little:

[image]

mck1117, 2021-12-06:
So standby mode does indeed need the WKUP pin wired - but we can test on a nucleo/discovery board.  They have jumpers you can remove to measure supply current on the CPU supply rail.

mck1117, 2021-12-06:
[image]

rusefillc, 2021-12-10:
@NMSTEC @mck1117 what's the plan? @NMSTEC says he has to cut a trace used by PPS2 on his Proteus for real test? what about MAX9924 how is that relevant how do we want to address it? I guess vr_descrete anyway

which LED is mentioned why and how?

why is CAN mentioned on the report? what is the CAN plan?

rusefillc, 2021-12-13:
https://github.com/rusefi/rusefi/wiki/Lua-Scripting#mcu_stop added

rusefillc, 2021-12-17:
@mck1117 @NMSTEC do you guys have a plan on what to cut on Proteus how?

rusefillc, 2021-12-25:
E65 BMW and #1731 are both saying 'hi'

rusefillc, 2021-12-26:
#3720 created

rusefillc, 2021-12-27:
https://github.com/andreika-git/hellen-one/issues/125 created

rusefillc, 2021-12-31:
Alex mentions TJA1054

rusefillc, 2021-12-31:
MC33742PEPR2
https://www.rocelec.com/part/FREMC33742PEPR2
[image]

rusefillc, 2021-12-31:
also there is the whole https://www.mouser.com/new/nxp-semiconductors/freescale-mc3490x-power-management/

rusefillc, 2022-01-01:
https://www.can-cia.org/fileadmin/resources/documents/proceedings/2017_hell.pdf

[image]

rusefillc, 2022-04-30:
Alphax 2chan has van wake up working

closed 2022-04-30 by rusefillc