rusefi/rusefi #3665: Pause CAN broadcast on engine stop

rusefillc, 2021-12-06:
Mark says that rusEFI CAN broadcast which is probably related to WBO control is keeping vehicle alive on engine stop

We need some way to stop WBO broadcast

Is 501 a relevant message?

NMSTEC, 2021-12-06:
Transmission of any data on Vehicle bus, wakes up BCM, and forces it to send GO SLEEP signal constantly with address 0x501 Packet 0x00.
In this specific case, 0ef50000 is the address in question. 

My recommendation is some sort of switch in FW that I can trip in LUA to enable and disable native can broadcast. I have already implemented a sort of graceful shutdown in LUA.

mck1117, 2021-12-06:
I don't think you want it on engine stop - you want it when the ignition is off.

rusefillc, 2021-12-06:
@mck1117 it sounds like we just need _some_ mechanism and lua hook to begin with?

Shall I simply add enableBroadcast() lua hook?

mck1117, 2021-12-06:
I would prefer `enableCanTx(bool)` that enables/disables _all_ CAN transmission

NMSTEC, 2021-12-06:
I'd request just all proteus side can to stop talking, lua controlled. I'd
like to keep full can control somewhere where I can easily access it

rusefillc, 2021-12-06:
@NMSTEC I am not sure what you are calling. Also we will absolutely not implement proteus-specific logic. For firmware issues it's all rusEFI.

NMSTEC, 2021-12-06:
> @NMSTEC <https://github.com/NMSTEC> I am not sure what you are asking.

Sorry, typing on phone while doing renovations. No, I am asking for a
switch that is accessed by lua to turn off and on rusefi broadcast, via
lua.

rusefillc, 2021-12-06:
@NMSTEC please elaborate why do you ask not the same what @mck1117 is asking

github comments are not slack. Please consider not using it while distracted while on the phone unless it's worth poking all subscribers via email?

mck1117, 2021-12-06:
he's asking for the same thing I did

rusefillc, 2021-12-07:
``enableCanTx(isEnabled)`` added
enabled by default

use enableCanTx(false) to supress CAN TX

closed 2021-12-07 by rusefillc