rusefi/rusefi #3676: Set sensors via LUA

NMSTEC, 2021-12-08:
Clutch and Brake switch are both canbus based on Harley. Some VAGs and GM also use brake switch via BCM.

Please allow to set brake on/off & clutch on/off in LUA.

NMSTEC, 2021-12-08:
Also would like to add control over redline, or have control over launch control via LUA.

closed 2022-01-04 by NMSTEC

rusefillc, 2022-01-04:
@NMSTEC why close this ticket?

NMSTEC, 2022-01-05:
Didn't this get solved? I have now been able to set sensors and switches via LUA from canbus.

rusefillc, 2022-01-05:
@NMSTEC please paste an example of Lua code which sets sensors/switches via Lua here

NMSTEC, 2022-01-05:
setCalibration(name, value, needEvent)

Sets specified calibration setting to specified value. Fires calibration
change event depending on needEvent parameter.

For example setCalibration("cranking.rpm", 900, false)

setClutchUpState(value)

<https://github.com/rusefi/rusefi/wiki/Lua-Scripting#setbrakepedalstatevalue>
setBrakePedalState(value)