rusefi/rusefi #3680: Throttle synchronization

rusefillc, 2021-12-09:
Zeiss says:

```
Throttle synchronization: this is now only the thought process of me, I do not know if it is 100% correct. The M70 has two independent throttles, each with its own TPS. Via the TPS Proteus knows how far the ETB are open, but only approximately. This means that if the values (angle to resistance) differ from each other, the ETB are opened differently. Correct? Via the ClosedLoop, the injection quantity is adapted to the air quantity, in each case per bank. Only it can be that one bank delivers more power than the other, because one bank gets more air (and as a consequence also gasoline) than the other. This is the reason why the two ETBs have to be synchronized against each other. It is not enough to calibrate them against the end stops, they must be balanced against each other. When they are properly balanced, not only is the idle much better and more stable, but so is throttle response and power.

The strategy here would be to measure the air mass (=MAF) or pressure (=MAP) and then open or close the ETB until the values match. BMW controls the ETB in the opposite direction during the adjustment. For this we need the evaluation for the second MAF or MAP sensor. I personally would prefer MAF, a normal analog one, not a digital one. After all, we already have the implementation for MAF/MAP, you just need to duplicate it and implement the algorithm for synchronization. It would make not only me happy, but everyone who has a V-motor with two ETBs.
```

billstewartx, 2022-01-05:
Jaguar V12 from 1995 also has 2 MAP sensors and 2 throttle bodies. So it would also benefit from this sync update.  I have a V12 setup  for a test mule.

HeinrichG-V12, 2022-02-15:
I've found so interesting infos how BMW syncronized the TBs:

1. get air mass from MAF (MAF because those engines use MAF sensors)
2. calculate injection amount (get this value from fuel map, "injection amount Y for air mass X")
3. calculate ti-time from injection amount (ti-time is injection duration or injector OPEN time)
4. use ti-time for TB sync

I think it absolutely doesn't matter if you have a MAP or MAF, the most important thing is to have one ti-signal per bank and compare these two signals to synchronize the TBs. In the end you don't synchronize the opening angles but the throughput.

mck1117, 2022-02-15:
> use ti-time for TB sync

this is a vestige of how the computer was built (easier to compute with injection time than fuel quantity), but we probably don't want to do that. We live in a world now where the fuel _pressure_ might be different per bank (dual bank GDI, anyone?), so the injection time might be different for each bank to deliver the same fuel.

HeinrichG-V12, 2022-02-15:
The fuel pressure should always be very similar across the banks, if it is not the case, you have there so problems.... I have observed it on my N73 (GDI engine), the pressure in the left and right bank was very close. With the intake manifold injectors one has (usually) no pressure sensors or am I wrong?

How could you implement the TB sync? Simply via the air mass (MAF) or negative pressure (MAP)?

mck1117, 2022-02-15:
I'd balance airmass, not airflow or anything about injection.

HeinrichG-V12, 2022-02-15:
Sound good for me! :)

HeinrichG-V12, 2022-10-16:
Hello Matthew,

any progress here?

Many thanks & regards,
Heinrich

billstewartx, 2022-10-16:
Heinrich... you have a N73?  me too! is yours in a 760LI?

rusefillc, 2022-10-16:
@mck1117 :)

HeinrichG-V12, 2022-10-18:
> Heinrich... you have a N73? me too! is yours in a 760LI?

Yes, I have a N73, installed in a 09/2006 760Li :)

billstewartx, 2022-10-18:
I feel your PAIN!

rusefillc, 2022-10-18:
While @HeinrichG-V12 has N73 in his life I believe it's the nice daily vehicle not the v12 in need of this throttle synchonization ticket?

HeinrichG-V12, 2022-10-18:
@billstewartx Pain? Why? The E65 is a marvelous daily driver. This ticket, for throttle sync, is for my M70, installed in 8 series.

mck1117, 2022-10-18:
Bill's E65 has a long sequence of chronic electrical issues. They're a great car when they work, but if things go seriously wrong it's BAD.

HeinrichG-V12, 2023-03-12:
@mck1117 

Hello Matthew,

some progress here?

Regards,
Heinrich

rusefillc, 2026-11-15:
https://rusefi.com/forum/viewtopic.php?t=3177

bernatsylvestre-boop, 2026-11-15:
[EN] Another use case: twin-cylinder engine with individual throttle bodies, one
ETB and one MAP sensor per cylinder. Balancing at idle/light load by equalizing
averaged runner MAPs (fixed crank-angle window), with the correction frozen
above a TPS threshold and learned per RPM x TPS cell. Happy to test.

[FR] Autre cas d'usage : moteur bicylindre à papillons individuels, avec un
papillon motorisé et un capteur MAP par cylindre. Équilibrage au ralenti et à
faible charge en égalisant les pressions moyennes des pipes (fenêtre angulaire
fixe du vilebrequin), avec une correction figée au-delà d'un seuil de TPS et
apprise par case régime x TPS. Je suis disponible pour tester.

bernatsylvestre-boop, 2026-11-16:
Follow-up after reading the code: setEtbAdd() calls setEtbLuaAdjustment(),
which loops over ETB1 and ETB2 and applies the same value to both, so Lua
cannot trim one throttle against the other. Each EtbController already has
its own luaAdjustment (exposed as etb1luaAdjustment / etb2luaAdjustment),
so the change is small: an optional second argument,
setEtbAdd(percent, etbNumber), with etbNumber = 1 or 2 and the current
behavior kept when omitted. I'll open a PR with this once I've tested it on
my Super-uaEFI.